G6EJD/ESP32-e-Paper-Weather-Display #130: 4.2 Version - Moon symbol on snow

bobybobybob, 2020-12-30:
on 4.2 Version on snow symbol, the moon is to much up

[url=https://postimg.cc/0684KDKC][img]https://i.postimg.cc/0684KDKC/snow.jpg[/img][/url]

bobybobybob, 2020-12-30:
https://www.bilder-upload.eu/bild-2ee4fd-1609364266.jpg.html

bobybobybob, 2020-12-30:
other symbol + moon look very ok  https://www.bilder-upload.eu/bild-b8f9b5-1609364390.jpg.html

G6EJD, 2020-12-30:
Try the update

bobybobybob, 2020-12-31:
Dont know if it solve problem... after update some libaries it wont compile. much red for U8g2_for_Adafruit_GFX.cpp

G6EJD, 2020-12-31:
They updated that library today

bobybobybob, 2021-01-01:
OK after fresh new install of Arduino IDE + Other, compiling work and upload too.

But now

Connecting to: PowerWLAN
WiFi connected at: 192.168.1.243
Busy Timeout!
Entering 654-secs of sleep time
Awake for : 10.596-secs

only one time at 1PM it got Weaterbrodcast... other only Bussy Timeout.   Better i do no updates anymore LOL

G6EJD, 2021-01-01:
What is the serial port /monitor saying? Are you receiving weather data?

bobybobybob, 2021-01-02:
Dont know why... now it work again. 

+5 is not good,   +10 is nice     see photographics picture

https://www.bilder-upload.eu/bild-6830e8-1609610493.jpg.html

yes very nice and thank you for helping to find the right parameter to test and find good number. :-)

G6EJD, 2021-01-02:
I’ve made that change now thank you.

closed 2021-01-02 by G6EJD

bobybobybob, 2021-01-13:
New Moon Problems...   

Big Symbol moon too high   and little 21 oClock symbol is moon too low?

https://www.bilder-upload.eu/bild-ee1eb5-1610568436.jpg.html

G6EJD, 2021-01-13:
Fixed that

bobybobybob, 2021-01-13:
big symbol is rain not snow

G6EJD, 2021-01-14:
Try that