G6EJD/ESP32-e-Paper-Weather-Display #132: Support for HD screens

madnat, 2021-01-03:
Hi there

Great project - thanks for sharing.  Just checking of there are any plans to support the HD versions of the Waveshare ePaper screens like this one: https://www.waveshare.com/product/displays/e-paper/epaper-1/7.8inch-e-paper-hat.htm - 1872x1404 would produce the space and resolution to have some wonderful content.

Thanks again

G6EJD, 2021-01-03:
The support for that display is dependent on the GxEPD2 library and until he gets a display support won’t be provided, sorry.

closed 2021-01-03 by G6EJD

madnat, 2021-01-04:
Thanks for the quick reply - I do see that in release 1.2.14 of the driver added support for the 6" 1448x1072 ED060KC1 on Waveshare e-Paper IT8951 Driver HAT.  Would your application drive that display?  Thank you

G6EJD, 2021-01-04:
Yes the application will support that screen size, albeit the display area is significantly larger and so would need to be either re-sized or the layout changed to accommodate the extra space.