G6EJD/ESP32-e-Paper-Weather-Display #133: 2.9 e-paper b/w/r

florin03, 2021-01-15:
Hello, 
Can you please add to code -> 2.9 e-paper b/w/r, this is the one from GxEPD library : 
GxGDEW029Z10/GxGDEW029Z10.h    // 2.9" b/w/r
Thank You

G6EJD, 2021-01-15:
I could do, but you only need to change 1-line for your display that’s it. Change the colours to your choice.

florin03, 2021-01-18:
Hi,
Maybe you can give me more info on what to change, i replaced in your code the line :
GxEPD2_BW<GxEPD2_290, GxEPD2_290::HEIGHT> display(GxEPD2_290(/*CS=D8*/ EPD_CS, /*DC=D3*/ EPD_DC, /*RST=D4*/ EPD_RST, /*BUSY=D2*/ EPD_BUSY));

with

GxEPD2_3C<GxEPD2_290c, GxEPD2_290c::HEIGHT> display(GxEPD2_290c(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4));

from GxEPD2 exemple, just nothing apears on the screen.
Thank You

florin03, 2021-01-18:
// BUSY -> 4, RST -> 16, DC -> 17, CS -> SS(5), CLK -> SCK(18), DIN -> MOSI(23), GND -> GND, 3.3V -> 3.3V

G6EJD, 2021-01-18:
OK all looks in order, so there are different display types please post a link to the product you have

florin03, 2021-01-18:
https://www.aliexpress.com/item/4001338269518.html?spm=a2g0s.9042311.0.0.71f94c4drYlyX1

G6EJD, 2021-01-18:
OK, looks suitable, so now what ESP32 board are you using link please.

florin03, 2021-01-18:
atm i am usig this one: 
https://www.aliexpress.com/item/1005001413672426.html?spm=a2g0s.9042311.0.0.71f94c4drYlyX1

it works with GxEPD2 exemple scketch, just not with your software.

florin03, 2021-01-18:
I managed to make it work by selecting board wemos lolin32 and changing info from own.credentials.h to my wifi/contry/city..... maybe it need all this info for software to run.
thanks for help

G6EJD, 2021-01-18:
OK, but the GxEPD2 has been changed a lot, look at the screen initiation section of the 7.5” version and add that line I’ll look it up latter.

G6EJD, 2021-01-18:
display.init(115200, true, 2);

G6EJD, 2021-01-18:
Sou saline it’s running now, I assumed you had set the WiFi ssid and password and API key

closed 2021-01-18 by G6EJD

florin03, 2021-01-18:
Yes it works, i presumed that will ~work without completing info in own.credentials.h, and i was wrong :)

G6EJD, 2021-01-18:
You need to set each line of the credentials, WiFi, country code, location, timezone, then adjust the colours as required.