G6EJD/ESP32-e-Paper-Weather-Display #134: Rainfall + Snow

bobybobybob, 2021-01-17:
Hello,
it look like that display show only real Rainfall but not Snowfall too, what is frozen rainfall. 

Make rain + snow to the same in graph and changing rainfall to precipitation that would make information more usefull

In Serial print output i see  rain + snow seperate. Maybee it would be best to use the bigest value of that two  or rain+Snow

G6EJD, 2021-01-17:
Some versions do, some don't depends on screen space, which version do you mean?

bobybobybob, 2021-01-17:
i have 4.2 display   
it would fine just rain+snow as one value in that graph. in detail is not so important for me because it often depends on the temperature. For details there are the three small icons in the upper right corner. I would like to see snow in the lower right corner, if it is raining or snowing. Pure indication that just what comes down from above.

bobybobybob, 2021-01-17:
https://www.bilder-upload.eu/bild-bb52da-1610885970.jpg.html   like example pic

G6EJD, 2021-01-20:
Perhaps propose a solution via a pull request

closed 2021-01-20 by G6EJD

bobybobybob, 2021-01-20:
will 

snowrain[0] = WxConditions[0].Rainfall + WxConditions[0].Snowfall;

work?  I not very programable coder

G6EJD, 2021-01-20:
Probably give it a try and see what it looks like, do they ever report both rain and snow? Otherwise the values could get too high.

bobybobybob, 2021-01-20:
Dont know..   here i see rain and snow 

https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/master/src/common.h

G6EJD, 2021-01-20:
This:
//#########################################################################################
void DisplayPrecipitationSection(int x, int y) {
  display.drawRect(x, y - 1, 167, 56, GxEPD_BLACK); // precipitation outline
  u8g2Fonts.setFont(u8g2_font_helvB10_tf);
  if (WxForecast[1].Rainfall > 0.005) { // Ignore small amounts
    drawString(x + 5, y + 15, String(WxForecast[1].Rainfall, 2) + (Units == "M" ? "mm" : "in"), LEFT); // Only display rainfall total today if > 0
    addraindrop(x + 65 - (Units == "I" ? 10 : 0), y + 16, 7);
  }
  if (WxForecast[1].Snowfall > 0.005)  // Ignore small amounts
    drawString(x + 5, y + 35, String(WxForecast[1].Snowfall, 2) + (Units == "M" ? "mm" : "in") + " * *", LEFT); // Only display snowfall total today if > 0
}

Becomes:
//#########################################################################################
void DisplayPrecipitationSection(int x, int y) {
  display.drawRect(x, y - 1, 167, 56, GxEPD_BLACK); // precipitation outline
  u8g2Fonts.setFont(u8g2_font_helvB10_tf);
  WxForecast[1].Rainfall += WxForecast[1].Snowfall;  // now the total is precipitation (rain + snow)
  if (WxForecast[1].Rainfall > 0.005) { // Ignore small amounts
    drawString(x + 5, y + 15, String(WxForecast[1].Rainfall, 2) + (Units == "M" ? "mm" : "in"), LEFT); 
    addraindrop(x + 65 - (Units == "I" ? 10 : 0), y + 16, 7);
  }
}

bobybobybob, 2021-01-27:
Thank you. I will try it and give feedback if it is working well.

bobybobybob, 2021-02-07:
That what look working is at 

void DrawForecastSection(int x, int y) {
++++
rain_readings[r]  = WxForecast[r].Rainfall + WxForecast[r].Snowfall;

In the serial print output data, it look like that there is only snow or rain and not both at same time. 

https://www.bilder-upload.eu/bild-c2122e-1612732790.jpg.html

G6EJD, 2021-02-07:
Can you copy the serial port output when this occurs please, scales are different but the sum should be totalised result.

bobybobybob, 2021-02-07:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

Connecting to: i-like-computers
WiFi connected at: 192.168.1.238

Creating object...and  Decoding weather data
 Lon: 13.51
 Lat: 51.60
Main: Snow
For0: Schnee

Icon: 13n
Temp: 0.87
Pres: 997.00
Humi: 79.00
TLow: 0.00
THig: 1.67
WSpd: 4.00
WDir: 252.00
CCov: 100
Visi: 241
Rain: 0.00
Snow: 0.89
Ctry: DE
SRis: 1612679823
SSet: 1612715225
TZon: 3600

Creating object...and  Decoding forecast data

Receiving Forecast period - 
Period-0--------------
DTim: 1612731600
Temp: 0.59
TLow: 0.30
THig: 0.59
Pres: 997.00
Humi: 88.00
For0: Snow

Icon: 13n
Desc: Schnee
CCov: 100
WSpd: 2.84
WDir: 249.00
Rain: 0.00
Snow: 3.89
Pop:  0.94
Peri: 2021-02-07 21:00:00

Period-1--------------
DTim: 1612742400
Temp: 0.08
TLow: -0.14
THig: 0.08
Pres: 998.00
Humi: 94.00
For0: Snow
[... 410 lines cut ...]
Period-23--------------
DTim: 1612980000
Temp: -3.76
TLow: -3.76
THig: -3.76
Pres: 1008.00
Humi: 98.00
For0: Snow

Icon: 13n
Desc: Schnee
CCov: 100
WSpd: 2.51
WDir: 344.00
Rain: 0.00
Snow: 3.66
Pop:  1.00
Peri: 2021-02-10 18:00:00
13n
13n
13n
04n
Entering 955-secs of sleep time
Awake for : 4.881-secs
Starting deep-sleep period...

G6EJD, 2021-02-08:
What did you get for the same period?

Regards

David

From: bobybobybob
Sent: 07 February 2021 21:41
To: G6EJD/ESP32-e-Paper-Weather-Display
Cc: G6EJD; State change
Subject: Re: [G6EJD/ESP32-e-Paper-Weather-Display] Rainfall + Snow (#134)

ets Jun 8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Connecting to: i-like-computers
WiFi connected at: 192.168.1.238
Creating object...and Decoding weather data
Lon: 13.51
Lat: 51.60
Main: Snow
For0: Schnee
Icon: 13n
Temp: 0.87
Pres: 997.00
Humi: 79.00
TLow: 0.00
THig: 1.67
WSpd: 4.00
WDir: 252.00
CCov: 100
Visi: 241
Rain: 0.00
Snow: 0.89
Ctry: DE
SRis: 1612679823
SSet: 1612715225
TZon: 3600
Creating object...and Decoding forecast data
Receiving Forecast period -
Period-0--------------
DTim: 1612731600
Temp: 0.59
TLow: 0.30
THig: 0.59
Pres: 997.00
Humi: 88.00
For0: Snow
Icon: 13n
Desc: Schnee
CCov: 100
WSpd: 2.84
WDir: 249.00
Rain: 0.00
Snow: 3.89
Pop: 0.94
Peri: 2021-02-07 21:00:00
Period-1--------------
DTim: 1612742400
Temp: 0.08
TLow: -0.14
[... 369 lines cut ...]
Peri: 2021-02-10 15:00:00
Period-23--------------
DTim: 1612980000
Temp: -3.76
TLow: -3.76
THig: -3.76
Pres: 1008.00
Humi: 98.00
For0: Snow
Icon: 13n
Desc: Schnee
CCov: 100
WSpd: 2.51
WDir: 344.00
Rain: 0.00
Snow: 3.66
Pop: 1.00
Peri: 2021-02-10 18:00:00
13n
13n
13n
04n
Entering 955-secs of sleep time
Awake for : 4.881-secs
Starting deep-sleep period...

bobybobybob, 2021-02-14:
((What did you get for the same period? ))    Hmm... what do you mean?

Current i am very happy with that solution. addition snow and rain in one bar graphic.

G6EJD, 2021-02-14:
I thought from the context that you were getting the wrong result.

bobybobybob, 2021-02-27:
After 2,5 week testing, snow+rain in one graph is working full wonderful and i will keep that on my weater epaper devices.