G6EJD/ESP32-e-Paper-Weather-Display #135: Feature request : Integrate Lilygo 4.7inch epaper module

lbonherbe, 2021-02-02:
Hello All!

Lilygo has recently released a 4.7inch epaper module, which is cheap, ESP32 integrated and has a great resolution of 960*540. It would be great to have this software working on it. I tried a bit to have it done but to no avail so far - it is way beyond my own coding capabilities!

G6EJD, 2021-02-02:
send me an email via g6ejd.dynu.com or www.dsbird.org.uk then contact

lyusupov, 2021-02-27:
Weather sketch was there, but LilyGO staff has removed it due to unknown reasons:
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/commit/75380f53c3f01a0bc58e705d43d4a14bcf0dc9ab

G6EJD, 2021-02-27:
They removed it because of my copyright, it's here now: 
https://github.com/G6EJD/LilyGo-EPD-4-7-OWM-Weather-Display

closed 2021-02-27 by G6EJD

guyou, 2021-04-25:
> They removed it because of my copyright, it's here now:
> https://github.com/G6EJD/LilyGo-EPD-4-7-OWM-Weather-Display

This repository is closed now. What happens?

lyusupov, 2021-04-25:
Looks like it was just a name change. New one:
https://github.com/G6EJD/LilyGo-EPD-47-OWM-Weather-Display
Old variant:
https://github.com/G6EJD/LilyGo-EPD-4-7-OWM-Weather-Display-1