G6EJD/ESP32-e-Paper-Weather-Display #139: Alignment problem

classic-audio, 2021-03-20:
Really a great project / code.
Works for me - although I have 2 minor points.
Display: 1.54 inch

1. alignment og the forecast hours on the display (3 hour forecast) - start at the mid of the screen
2. No data of the actual wind - showing 0 (zero) - I am getting data from open weather server in another application with correct data 

Where can I make these changes to have the correct data and display of data?

[image]

G6EJD, 2021-03-20:
You should remove these two lines:
  if (WxConditions[0].Forecast1 != "") Wx_Description += ", " + WxConditions[0].Forecast1;
  if (WxConditions[0].Forecast2 != "") Wx_Description += ", " + WxConditions[0].Forecast2;
And try replacing this line:
  drawString(x + SCREEN_WIDTH / 2, y + 3, String(ConvertUnixTime(WxForecast[index].Dt + WxConditions[0].Timezone).substring(0,5)), CENTER);

with:
  drawString(x + offset, y + 3, String(ConvertUnixTime(WxForecast[index].Dt + WxConditions[0].Timezone).substring(0,5)), CENTER);

classic-audio, 2021-03-20:
I have a solution now - thank you very much for the fast reply.
The line:
drawString(x + offset, y + 3, String(ConvertUnixTime(WxForecast[index].Dt + WxConditions[0].Timezone).substring(0,5)), CENTER);
I made a small correction
drawString(x + offset / 2, y + 3, String(ConvertUnixTime(WxForecast[index].Dt + WxConditions[0].Timezone).substring(0,5)), CENTER);

Another small issue - at the right of the display - the boxes are missing lines at the right.
Perhaps you have a hint for that too ?

[image]

classic-audio, 2021-03-20:
For your info...
I also changed the text line of forecast to 2 lines - more easy to read :)

G6EJD, 2021-03-20:
Good to hear,

try:

  display.drawRect(x, y, offset + 1, 65, GxEPD_BLACK);

to:

Regards

David

From: Frede Kristensen
Sent: 20 March 2021 15:55
To: G6EJD/ESP32-e-Paper-Weather-Display
Cc: G6EJD; Comment
Subject: Re: [G6EJD/ESP32-e-Paper-Weather-Display] Alignment problem (#139)

For your info...
I also changed the text line of forecast to 2 lines - more easy to read :)

G6EJD, 2021-03-20:
Sorry,

Keyboard error:

Try:
  display.drawRect(x, y, offset + 1, 65, GxEPD_BLACK);
to:
  display.drawRect(x, y, offset , 65, GxEPD_BLACK);

Out of interest which time zone are you in?

Regards

David

From: Frede Kristensen
Sent: 20 March 2021 15:55
To: G6EJD/ESP32-e-Paper-Weather-Display
Cc: G6EJD; Comment
Subject: Re: [G6EJD/ESP32-e-Paper-Weather-Display] Alignment problem (#139)

For your info...
I also changed the text line of forecast to 2 lines - more easy to read :)

closed 2021-03-20 by G6EJD

classic-audio, 2021-03-20:
Thank you very much - fast reply and it works.

I am in Europe - Copenhagen time zone.
I live on a small Island "Venø" - central Jytland.