G6EJD/ESP32-e-Paper-Weather-Display #140: Problems with using fonts with polish symbols.

jacekpoz, 2021-03-21:
I have a problem with using fonts that aren't in the project. I want to get polish symbols, I tried unifont for u8g2, I also converted the polish .ttf fonts on [this](https://rop.nl/truetype2gfx/) website. Then I tried fontconvert from Adafruit GFX and it doesn't work either. The project compiled, but the display doesn't show any text. How would I fix that?

G6EJD, 2021-03-21:
The fonts need to be in GFX format I suggest you go back to a working font and then try one of the many font editors that can produce GFX fonts but only try one character at a time to begin with. I have that with the ° symbol and it works you will need to use or amend existing character positions below 127 eg change u to ū or whatever

jacekpoz, 2021-03-21:
Which editors would you suggest using?

G6EJD, 2021-03-21:
https://tchapi.github.io/Adafruit-GFX-Font-Customiser/

jacekpoz, 2021-03-21:
I found the issue and managed to fix the problem, thanks a lot for the help.

closed 2021-03-21 by jacekpoz