G6EJD/ESP32-e-Paper-Weather-Display #141: i keep getting this error message when i try and upload is there a way to fix it

jobbo21, 2021-03-22:
Arduino: 1.8.13 (Windows 10), Board: "TTGO T1, Huge APP (3MB No OTA/1MB SPIFFS), 240MHz (WiFi/BT), DIO, 40MHz, 4MB (32Mb), 115200, None"

In file included from C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson.hpp:30:0,

                 from C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson.h:9,

                 from C:\Users\08015961\Documents\Arduino\Waveshare_2_13_T5\Waveshare_2_13_T5123\Waveshare_2_13_T5123.ino:22:

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/ConverterImpl.hpp: In instantiation of 'static T ArduinoJson6173_91::Converter<T, Enable>::fromJson(ArduinoJson6173_91::VariantConstRef) [with T = char*; Enable = void]':

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/VariantRef.hpp:98:34:   required from 'T ArduinoJson6173_91::VariantRef::as() const [with T = char*]'

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Object/MemberProxy.hpp:86:52:   required from 'TValue ArduinoJson6173_91::MemberProxy<TParent, TStringRef>::as() const [with TValue = char*; TObject = ArduinoJson6173_91::ElementProxy<ArduinoJson6173_91::MemberProxy<ArduinoJson6173_91::ObjectRef, const char*> >; TStringRef = const char*]'

C:\Users\08015961\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display-master\src/common.h:39:72:   required from here

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/ConverterImpl.hpp:24:20: error: no matching function for call to 'convertFromJson(char*&, ArduinoJson6173_91::VariantConstRef&)'

     convertFromJson(value, variant);  // Error here? See https://arduinojson.org/v6/unsupported-as/

                    ^

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/ConverterImpl.hpp:149:62: note: candidate: template<class T> typename ArduinoJson6173_91::enable_if<ArduinoJson6173_91::IsWriteableString< <template-parameter-1-1> >::value>::type ArduinoJson6173_91::convertFromJson(T&, ArduinoJson6173_91::VariantConstRef)

 inline typename enable_if<IsWriteableString<T>::value>::type convertFromJson(

                                                              ^

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/ConverterImpl.hpp:149:62: note:   template argument deduction/substitution failed:

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/ConverterImpl.hpp: In substitution of 'template<class T> typename ArduinoJson6173_91::enable_if<ArduinoJson6173_91::IsWriteableString< <template-parameter-1-1> >::value>::type ArduinoJson6173_91::convertFromJson(T&, ArduinoJson6173_91::VariantConstRef) [with T = char*]':

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/ConverterImpl.hpp:24:20:   required from 'static T ArduinoJson6173_91::Converter<T, Enable>::fromJson(ArduinoJson6173_91::VariantConstRef) [with T = char*; Enable = void]'

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/VariantRef.hpp:98:34:   required from 'T ArduinoJson6173_91::VariantRef::as() const [with T = char*]'

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Object/MemberProxy.hpp:86:52:   required from 'TValue ArduinoJson6173_91::MemberProxy<TParent, TStringRef>::as() const [with TValue = char*; TObject = ArduinoJson6173_91::ElementProxy<ArduinoJson6173_91::MemberProxy<ArduinoJson6173_91::ObjectRef, const char*> >; TStringRef = const char*]'

C:\Users\08015961\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display-master\src/common.h:39:72:   required from here

C:\Users\08015961\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/ConverterImpl.hpp:149:62: error: no type named 'type' in 'struct ArduinoJson6173_91::enable_if<false, void>'

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\08015961\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

Error compiling for board TTGO T1.

jobbo21, 2021-03-22:
I'll check I've get the latest library, I'm trying to use the t5 2.13

G6EJD, 2021-03-22:
Th code is ok, all your error messages were associated with the Arduino JSON library

jobbo21, 2021-03-23:
I've got the latest library and still having the same issue

G6EJD, 2021-03-23:
You say you can’t upload, do you mean it won’t compile?
What board type are you selecting?

G6EJD, 2021-03-23:
Do have any other boards to try?

jobbo21, 2021-03-23:
I've been playing about with it now the only error I'm getting is this
In file included from C:\Users\08015961\AppData\Local\Temp\arduino_build_444196\sketch\src\GxGDEH0213B72\GxGDEH0213B72.cpp:14:0:
C:\Users\08015961\AppData\Local\Temp\arduino_build_444196\sketch\src\GxGDEH0213B72\GxGDEH0213B72.h:17:22: fatal error: ../GxEPD.h: No such file or directory
compilation terminated.

Schramme793, 2021-03-23:
Arduino: 1.8.13 (Mac OS X), Board: "WEMOS LOLIN32, 80MHz, Default, 240MHz (WiFi/BT), 921600"

In file included from /Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantRef.hpp:14:0,
                 from /Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from /Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from /Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson.hpp:17,
                 from /Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson.h:9,
                 from /Users/christianschramm/Desktop/ESP32-e-Paper-Weather-Display-master/examples/Waveshare_2_9/Waveshare_2_9.ino:24:
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAs.hpp: In instantiation of 'T ArduinoJson6173_91::variantAs(ArduinoJson6173_91::VariantData*, ArduinoJson6173_91::MemoryPool*) [with T = char*]':
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantRef.hpp:196:24:   required from 'T ArduinoJson6173_91::VariantRef::as() const [with T = char*]'
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Object/MemberProxy.hpp:86:52:   required from 'TValue ArduinoJson6173_91::MemberProxy<TParent, TStringRef>::as() const [with TValue = char*; TObject = ArduinoJson6173_91::ElementProxy<ArduinoJson6173_91::MemberProxy<ArduinoJson6173_91::ObjectRef, const char*> >; TStringRef = const char*]'
/Users/christianschramm/Documents/Arduino/libraries/ESP32-e-Paper-Weather-Display-master/src/common.h:39:72:   required from here
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAs.hpp:57:22: error: no matching function for call to 'variantAs(ArduinoJson6173_91::VariantData*&)'
   return variantAs<T>(data);
                      ^
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAs.hpp:23:1: note: candidate: template<class T> typename ArduinoJson6173_91::enable_if<((ArduinoJson6173_91::is_integral<T>::value && (! ArduinoJson6173_91::is_same<bool, T>::value)) && (! ArduinoJson6173_91::is_same<char, T>::value)), T>::type ArduinoJson6173_91::variantAs(const ArduinoJson6173_91::VariantData*)
 variantAs(const VariantData* data) {
 ^
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAs.hpp:23:1: note:   template argument deduction/substitution failed:
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAs.hpp: In substitution of 'template<class T> typename ArduinoJson6173_91::enable_if<((ArduinoJson6173_91::is_integral<T>::value && (! ArduinoJson6173_91::is_same<bool, T>::value)) && (! ArduinoJson6173_91::is_same<char, T>::value)), T>::type ArduinoJson6173_91::variantAs(const ArduinoJson6173_91::VariantData*) [with T = char*]':
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAs.hpp:57:22:   required from 'T ArduinoJson6173_91::variantAs(ArduinoJson6173_91::VariantData*, ArduinoJson6173_91::MemoryPool*) [with T = char*]'
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantRef.hpp:196:24:   required from 'T ArduinoJson6173_91::VariantRef::as() const [with T = char*]'
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Object/MemberProxy.hpp:86:52:   required from 'TValue ArduinoJson6173_91::MemberProxy<TParent, TStringRef>::as() const [with TValue = char*; TObject = ArduinoJson6173_91::ElementProxy<ArduinoJson6173_91::MemberProxy<ArduinoJson6173_91::ObjectRef, const char*> >; TStringRef = const char*]'
/Users/christianschramm/Documents/Arduino/libraries/ESP32-e-Paper-Weather-Display-master/src/common.h:39:72:   required from here
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAs.hpp:23:1: error: no type named 'type' in 'struct ArduinoJson6173_91::enable_if<false, char*>'
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAs.hpp: In instantiation of 'T ArduinoJson6173_91::variantAs(ArduinoJson6173_91::VariantData*, ArduinoJson6173_91::MemoryPool*) [with T = char*]':
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantRef.hpp:196:24:   required from 'T ArduinoJson6173_91::VariantRef::as() const [with T = char*]'
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Object/MemberProxy.hpp:86:52:   required from 'TValue ArduinoJson6173_91::MemberProxy<TParent, TStringRef>::as() const [with TValue = char*; TObject = ArduinoJson6173_91::ElementProxy<ArduinoJson6173_91::MemberProxy<ArduinoJson6173_91::ObjectRef, const char*> >; TStringRef = const char*]'
/Users/christianschramm/Documents/Arduino/libraries/ESP32-e-Paper-Weather-Display-master/src/common.h:39:72:   required from here
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAs.hpp:29:55: note: candidate: template<class T> typename ArduinoJson6173_91::enable_if<ArduinoJson6173_91::is_enum<T>::value, T>::type ArduinoJson6173_91::variantAs(con [... cut ...] Proxy<ArduinoJson6173_91::MemberProxy<ArduinoJson6173_91::ObjectRef, const char*> >; TStringRef = const char*]'
/Users/christianschramm/Documents/Arduino/libraries/ESP32-e-Paper-Weather-Display-master/src/common.h:39:72:   required from here
/Users/christianschramm/Documents/Arduino/libraries/ArduinoJson-6-2.x/src/ArduinoJson/Variant/VariantAsImpl.hpp:14:70: error: no type named 'type' in 'struct ArduinoJson6173_91::enable_if<false, char*>'
exit status 1

Error compiling for board WEMOS LOLIN32.

Schramme793, 2021-03-23:
I'm having a quite similar issue with ArduinoJson library. I tried to complile the code for 2.9 inch Waveshare display on two different LOLIN32 boards with the same issues. I'm also using the latest ArduinoJson library from github.

I also tried to compile the boards with other example codes like WifiScan to test them. Compiling always worked.
Did this issue only belongs to the ArduinoJson library?

G6EJD, 2021-03-23:
It is much better giving me the standard rather than verbose error log from the compiler.
Try downgrade it the Arduino JSON to the previous version -2 
I’ve just taken the same code and it compiles perfectly so this does suggest your IDE installation has an error.
Have you tried compiling one of the Arduino JSON examples from File, Examples, Arduino JSON?

G6EJD, 2021-03-23:
@Schramme793 did you try downgrading the Arduino JSON library? 
Thanks for letting us know.
I’ll go through sone most examples

G6EJD, 2021-03-23:
Update:  I'm using Arduino IDE 1.8.13, ESP32 add-on 1.0.5, Arduino JSON 6.17.3 by Benoit Blanchon
As a sample, I've just compiled with no errors versions for 1.54", 2.9", 2.9" T5, 4.2", 7.5"
Are you sure your using the latest 'forecast_record.h' and 'common.h' - recently updated in the last few months?

closed 2021-03-23 by G6EJD

G6EJD, 2021-03-26:
I’m sure you used the wrong Arduino JSON library, have you got it fixed now?

Schramme793, 2021-03-26:
Yes, I downgraded the Arduino Json library to 6.17.2 and it worked. Thanks for your support!

G6EJD, 2021-03-26:
Ok thanks, I’ll check out what’s changed or wrong and feed that back. Up until now it’s been faultless.