G6EJD/ESP32-e-Paper-Weather-Display #142: Lolin32 + Waveshare 2.9 inch e-Paper -- wind direction and forecast issue

Schramme793, 2021-03-24:
[image]

My Weatherstation is finally working and I'm quite happy with result. But there are some aspects I would like to change. For the one thing the alphabetic wind direction isn't shown completely underneath the compass. For example the wind direction "WSW" is cut off, as you can see in the photo above. 
Unfortunately I'm not really sure which line in the code I have to adapt to fix the position of the wind direction. For another thing I would like to see the forecast for the next three days. As shown in the YouTube video (https://www.youtube.com/watch?v=S-rqia4tIn4&t=338s) I would like to display the forecast for the next three days instead of the forecast for the next 9 hours.
Is there a chance to switch this forecast mode?

Thanks for this great project and your support.

G6EJD, 2021-03-24:
Line 284 in the 2.9” version 
drawString(x - 2, y + Cradius + 10, WindDegToDirection(angle), CENTER);

To maybe:
drawString(x - 10, y + Cradius + 10, WindDegToDirection(angle), CENTER);

I’ll look at the forecast

Schramme793, 2021-03-26:
Thanks for your advice. Your propose for the change in the line for the wind direction works great. 
Have you an idea for the forecast?

G6EJD, 2021-03-26:
Yes I’m working on it, but not straightforward as the data is sent out at different times, for example say it was set in UTC time for the 0900 data for the next day it’s just a snap shot, and then for the next day 0900 and so on for the 3-days, but when I change the time to say Germany they forecast is still given at 0900 UTC so 1000 in the EU and in other places a different offset so I’m trying to figure out how or why OWM does that I thought for each location they would issue fixed forecast times of 00:00 03:00 06:00 09:00 and so on, but no some are 07:00 or 08:00 or 11:00 and coding for all those is not easy!

Schramme793, 2021-03-26:
Ok, that sounds quite complex. I didn’t expect the forecast to be such a problem. Sorry for that, but I‘m quite sure you can figure something out.

G6EJD, 2021-03-26:
I think it’s a good idea just got to work it out, on the larger displays it can do 3x3 hour forecasts then do 5-days of forecast, plus the expected hi/lo temperatures which I forgot to mention ad it’s no good having 08:00 temperatures for the forecast, I’m making it find the highest and lowest in the forecast day then to use those.

G6EJD, 2021-03-29:
Take a look at https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/tree/master/examples/Waveshare_2_9
I've fixed quite a lot, added 3 forecast boxes showing Hi/Lo for the next 3-days, detail has gone. You will need the forecast_record.h file to be updated if not already.
Please try it out and let me of any issues.

G6EJD, 2021-03-30:
Now added: https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/master/examples/Waveshare_2_9/Waveshare_2_9_v3.ino
Which corrected all the icons and ensured day-3 forecast was visible in the early morning.

SzymonSlupik, 2021-03-30:
Thanks for all this, great project!
BTW, does this also affect the
https://github.com/G6EJD/LilyGo-EPD-4-7-OWM-Weather-Display? Or is v2.72
already corrected?

G6EJD, 2021-03-30:
Well I have a later EPD4.7 version 3.4 that adds forecast days, PM if you want a copy.

G6EJD, 2021-03-30:
OK made those changes now, see here:
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/master/examples/Waveshare_2_9/Waveshare_2_9_v4.ino
I have removed the sunrise and sunset text and used symbols, I needed to do that to leave space for -ve temperatures and widened the forecast boxes to fit that range. please check it and then I think this can be closed.

G6EJD, 2021-03-31:
I’ll leave both versions there for a few weeks, then leave v4 as the final version. Thanks, I’m pleased it works ok.

closed 2021-03-31 by G6EJD