G6EJD/ESP32-e-Paper-Weather-Display #146: Question on how battery percentage is calculated

cgreening, 2021-06-06:
Not really an issue, more a question. I was wondering what the source for the battery percentage forumal is?

```
percentage = 2836.9625 * pow(voltage, 4) - 43987.4889 * pow(voltage, 3) + 255233.8134 * pow(voltage, 2) - 656689.7123 * voltage + 632041.7303; 
```

G6EJD, 2021-06-06:
It’s a 4th order polynomial, what I did is take the standard discharge curve of a Li-ion battery and overlaid that on a y-axis of 0-100% so that for a given voltage it would yield a % charge figure. The upper limit was 4.2v and lower 3.6v
I created the polynomial using the classic table method, you can find examples via google of how to create the factors, you need (I found) a lot of precision to get the desired result what I mean by that is the polynomial might be a good match but the input values would often give wacky results without the precision.
You could go for a simpler regime of any voltage above 3.7 is 100% and then have a small linear section between 3.7-3.5 where 3.5v is 0%.

Hope this helps.

closed 2021-06-06 by G6EJD

SzymonSlupik, 2021-06-07:
Thanks for the insights. There are circumstances when the calculations go wrong however... I have this regularly when the cell is 18650 and the level is close to discharged...
[image]

G6EJD, 2021-06-07:
Just change this line:
if (voltage <= 3.20) percentage = 0; 
to:
if (voltage <= 3.50) percentage = 0;

G6EJD, 2021-06-07:
I notice your forecast timezone are incorrect, not sure where you got the version from?

cgreening, 2021-06-07:
Very helpful! Reading around, battery percentage calculation seems to be a bit of an art form.

SzymonSlupik, 2021-06-07:
Thanks for the hint! I would argue though if 3.50 is really 0%, as it still keeps working :) Nevertheless, with this hint I can fine tune it to match my battery.

G6EJD, 2021-06-07:
As your using the LilyGo board, I don't know as the LDO regulator usually needs 0.15v so that's 3.3+0.15 = 3.45v hence 3.5v
Also Li-ion batteries lose their capacity when over discharged. 
I fitted an 18650 to my LilyGo board in Dec'20 and it's still working showing 67% after 7-months.

SzymonSlupik, 2021-06-07:
regarding the "where this version from" - have just checked the code and it says OWM_EPD47_epaper_v2.72

> // ESP32 Weather Display and a LilyGo EPD 4.7" Display, obtains Open Weather Map data, decodes and then displays it.
> // This software, the ideas and concepts is Copyright (c) David Bird 2021. All rights to this software are reserved.

You offered me an improved several months ago, but I never had the time to reach out back to you... Hope this is not the one that raises your blood pressure...

Back to the batteries - I must have really bad set of 18650s, as they barely last for two weeks... I have another unit with a 1550mAh Li-Poly and it has been running for months now. Do you have a [Joules] figure for a daily power budget? My goal is really to run this on supercaps and a solar cell, but my daily budget is below 10J (a small indoor panel). Have to do power profiling one day...

G6EJD, 2021-06-07:
My EPD consumes (44mA x 3 + 144 x 2) / 3600 = 422 / 3600 = 0.117mAh or at 3.3v = 1.37J 
I’m not used to using Joules only mAh

G6EJD, 2021-06-08:
> Well... I'm not entirely sure about your numbers... Perhaps you could disclose more about your hardware...
>
> I have just hooked a JouleScope to the TTGO T5 4.7" board (with display):
>
> It shows 270uA in deep sleep, which itself sums up to: 0.00027*3.3(v)*3600(s)*24(h) = 77J
> The startup sequence is about 3.5J
> Then the hourly update sequence is 7J *12=84J, bringing the total to ~160J.
>
> So from this perspective it is completely out of reach to have an indoor, photovoltaic - powered unit... :(
>
> (my measurements are by providing 3v3 to the output of the G3P LDO, so perhaps there is a reverse leak current somewhere... but I don't think this would change things materially...)

I’m getting similar results to you, my ucurrent gold gives a sleep current of 270uA too.
I have 1.5secs from start to finish of the startup and getting data from the OWM server at 144mA then 5.5secs at 44mA for the screen refresh so that’s (144x1.5.+44x5.5)/3600mAh=0.127mAh and 0.270x53/3600mAh= 0.004mAh for the sleep time. 7-secs on and 53 secs sleeping. 

Without converting to joules (x3.3vx3.6) a 2600mAh battery should last for 2600/(0.127 + 0.004)=19847 hours or 2.2 years and taking into account Battery degrading by 80% of the total and that yields 1.8 years and accords with what I’m seeing, it’s been running for 7-months with no discernible change in the battery voltage.

Regards

David

G6EJD, 2021-06-08:
A solarPV cell would easily drive your board and keep it charged, but the problem is however that power to the usb-c powers the board and does not allow it to sleep a s therein is the problem. It would be better to wire a solar pv cell across the battery terminals, then it would keep the battery charged.

Regards

David

SzymonSlupik, 2021-06-08:
Thanks for confirming the numbers, David!

I have done quite a lot with solarPV. It all depends on how much light
you're getting.

10J is my proven number for daily indoor (assuming worst case - short
winter overcast day). You're getting like 6 hours at 50 lux if you're
lucky. Outdoors (or even at a well exposed window) you may easily get
10-30x of that.

Also my idea has been to run it on super capacitors (unlimited lifetime)
rather than on Li-Poly (limited number of cycles). But again, a pair of 10F
capacitors (2.7V each) give in the order of 10J, not 100...

--Simon

G6EJD, 2021-06-08:
Simon,

The number joules in a 10F capacitor is:
Q=1/2CV^2
Therefore Q=1/2.10.3.3^2
Q=54Joules
Two in parallel would be 108joules

Regards

David

SzymonSlupik, 2021-06-08:
True.

But in practice the usable voltage range is 0.9V for two connected serially
(2 * 10F is 5F if you ask....) - from 4.5V (max charged) to 3.6V (cut-off).
So you have
- charged energy 5*4.5^2/2 = 50J
- cut-off energy 5*3.6^2/2 = 32J
- usable difference: 18J

The supercap voltage is very low - 2.7V, so need to connect serially...

-S.

G6EJD, 2021-06-08:
Understand, I didn’t know you have them in series. 

I’ve run a weather station off a solar 6v 1Watt panel for over 2-years. It provides more than enough energy to keep the battery charged. An 18650 single cell with the solar cell wired across it.

Regards

David

SzymonSlupik, 2021-06-10:
Yes, it all depends on the lighting conditions...

My goal was to use a panel similar in size to the display, but that
requires very strong light... in the order of 1000 lux.

BTW there are interesting developments on the supercapacitors front... a
new breed is now available with capacities of 100F in a package comparable
to 10F a year ago... I have ordered a couple to experiment with...

Will post some results if they are encouraging enough :)

S.

SzymonSlupik, 2021-07-30:
Hi David,

As promised - an update here - I've had this weather display running on
2x100F super capacitors, with the support from AEM SuCa board (
https://community.hiveeyes.org/t/aemsuca-solar-harvesting-into-supercapacitors/2382)
and a high-side CMOS power switch.

The key is the hysteresis - the power output is switched on at 3.9V and off
at 3.6V. When using 2x100F capacitor a single refresh cycle consumes ~0.1V,
so this means at 3.9V there is enough energy to execute three full refresh
cycles. I had tried this before with 2x10F capacitors and it did not work -
when they were charged to 3.9V, the board could not complete a single
refresh cycle before the voltage dropped to 3.6V and the power was cut off,
ending in an endless loop.

See the attached sketch.

S.