G6EJD/ESP32-e-Paper-Weather-Display #148: Occasional freeze when retrieving forecast json data

jhughes1010, 2021-06-28:
Occasionally, I notice the screen is blank when running. I initially thought that I might be reaching max attempts on data retrieval (nothing published to the screen on that branch) and modified to at least call DisplayGeneralInfoSection(); so I could see if things were alive and well, but that was not the problem.

I finally caught a situation where I was able to see on serial monitor, that forecast data was in the middle of period 21 and halted for some reason. Has anyone else seen this? I ended up making use of a WDT in the DecodeWeather() function. Someone else may have other ideas?

G6EJD, 2021-06-28:
Sometimes the volume of forecast data gets very big, especially if it’s reporting / forecast prolonged periods of rain, I suggest you increase the memory in common.h here:
DynamicJsonDocument doc(35 * 1024);
to:
DynamicJsonDocument doc(55 * 1024);

There is plenty of RAM to spare and the content of the API return has grown since this was first published.

jhughes1010, 2021-06-28:
Will do! Thanks for quick answer.

73 de KB0HHM

closed 2021-06-28 by jhughes1010

G6EJD, 2021-06-28:
Also, as all the forecasts are identical then a failure at record 21 suggests an out of memory problem as the process is repeated. Occasionally OWM has issues but these usually affect the whole message.
73 Dave

jhughes1010, 2021-06-28:
I increased buffer size as you suggested, but left my WDT code in place also. 

James Hughes