G6EJD/ESP32-e-Paper-Weather-Display #153: Fade image drawn on EPD

PaulGoldschmidt, 2021-08-04:
Hey,

after some trial-and-error, I got the software up and running. The ESP (Waveshare EPD - ESP32 Driver board) catches all the weather data and communicates with the display, so it refreshes successfully. The only problem is now that the drawn image has very low contrast and the image is very fade overall (while refreshing, the display shows for a short while the image with full contrast, other projects are able to draw images and text with the full contrast).

Any idea how to solve this?

Kind regards,
Paul

G6EJD, 2021-08-04:
Have you tried the GxEPD2 example to ensure it’s not the driver?
This is usually associated with the Busy connection being incorrect so that the display is not correctly drawn before drive is removed. Check the pin definitions for the Waveshare board are correct, I have a Waveshare board running without issues.

PaulGoldschmidt, 2021-08-04:
The GxEPD2-Example works fine (see [picture](https://ibb.co/x6fhrdQ)), the pins match the example.
[This is how](https://cloud.p3g3.de/s/gtGaB4wEQ2xDBYF) the refresh looks like with the weather program

PaulGoldschmidt, 2021-08-04:
I'm using the 7.5 inch B/W 800x480 px (V2) EPD by Waveshare

PaulGoldschmidt, 2021-08-04:
Everything should be up-to-date (using V16.11 for the Weather-Software)

G6EJD, 2021-08-04:
And the T7 version?

PaulGoldschmidt, 2021-08-04:
The newest one available￼ (if you could tell me where to look up the version I will do it asap)

G6EJD, 2021-08-04:
There are only two versions either the 7.5” or 7.5” T7

PaulGoldschmidt, 2021-08-04:
Got it, it is the T7-Version.

G6EJD, 2021-08-04:
[image]
Code works perfectly. I have just used the version here, compiled and uploaded to a Waveshare board connected to  aT7 display, no problems found. Your doing something wrong.
Most likely your display has been in direct sunlight and caused damage or it is too hot.
This weather display uses exactly the same screen drawing as the EDP2 example.

closed 2021-08-04 by G6EJD

stefan-sherwood, 2021-08-23:
For others experiencing similar symptoms, I am experienced a similar problem, which I filed details on [here](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/57). Using the Weather example [here](https://github.com/G6EJD/LilyGo-EPD-4-7-OWM-Weather-Display-1), this is what I am seeing. In real life it is quite faded, definitely looks better in the photo.

[image]

SzymonSlupik, 2021-08-24:
Looks like it was damaged by strong sunlight. This happened to me once, but
the display has recovered after about a day. Perhaps in your case the
damage is permanent.
I also had similar damage the other way around - dim background (see
attached - the display on the left).
They are fragile. Not sure why you can have an Amazon Kindle in full
sunshine but not a Waveshare display though...

stefan-sherwood, 2021-08-24:
I don't think it has to do with sunlight. I took it out of the package it shipped in, never exposed it to sunlight. It was working perfectly when I plugged and unplugged it several times. Then I uploaded the software and it was immediately faded.

PaulGoldschmidt, 2021-08-24:
Had the same issue and mine works perfectly using other libs, I also pretty sure that the issue is not caused by errors in handling.
Kind regards,
Paul

G6EJD, 2021-08-24:
I note the library was ‘updated’ in a bid to get a portrait mode, but it was not implemented correctly.

stefan-sherwood, 2021-08-25:
Unfortunately the GxEPD2 library doesn't support the board I'm using. I will continue to look for another that might. Thanks.

G6EJD, 2021-08-25:
You could go back a few versions on GitHub to the original version because that one worked.

stefan-sherwood, 2021-08-25:
Happy to try that. Which project (GxEPD2 or LilyGo-EPD47) are you referring to?

G6EJD, 2021-08-25:
LilyGo 4.7”

stefan-sherwood, 2021-08-25:
I tried going back to 8 different past version of LilyGo-EPD47 and they all produce the same low-contrast result. :(