G6EJD/ESP32-e-Paper-Weather-Display #155: Example for Waveshare 7.5" HD 880x528

CargoBikoMeter, 2021-08-25:
Hi,

please could you provide an example code for the Waveshare ePaper HD display HINK-E075A07-A0 with a resolution of 880x528? I changed the resolution in the Waveshare_7_5_T7 example from 800x480 to 880x528, but is does not work, the data are not displayed on the entire screen and some fields are not present. Only the  Waveshare_7-5 example shows all fields, but on a smaller area.

Currently I use the following display definition in combination with the following ESP32 board: 

GxEPD2_3C<GxEPD2_750c_Z90, GxEPD2_750c_Z90::HEIGHT / 2> display(GxEPD2_750c_Z90(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY)); // GDEH075Z90 880x528

ESP32 NodeMCU: https://www.reichelt.de/nodemcu-esp32-wifi-und-bluetooth-modul-debo-jt-esp32-p219897.html?&nbc=1

I bought this 3C display: https://www.reichelt.de/entwicklerboards-display-hd-epaper-7-5-schwarz-weiss-rot-debo-epa-7-5-rd-p253960.html?&nbc=1

Our plan is to combine the weather display application with an application to show the data from our Telraam device in Berlin (https://telraam.net/en/location/9000001786) and the PM-Sensor (https://opensensemap.org/explore/61115f385f5cac001b3bb4f2)

For more information see: 
 * https://adfc-tk.de/wir-zaehlen/
 * https://adfc-tk.de/2021/08/wir-messen-feinstaub-in-der-doerpfeldstrasse/

Regards,
Roland

G6EJD, 2021-08-25:
Did you adjust:
#define SCREEN_WIDTH  880             // Set for landscape mode
#define SCREEN_HEIGHT 528
If the display is updating correctly with the  750c_Z90 driver, then the task is to go through the code and move the x,y values of all print statements, but nonetheless they should still be within the boundaries of the 7.5" display.

I don't have one of those displays so I can't help further, sorry.

closed 2021-08-25 by G6EJD

GeorgH93, 2021-08-29:
Hi Roland

The display has to many pixel for the RAM of the ESP32, you have to use it in paged mode. You are already using the correct display definition. The `GxEPD2_750c_Z90::HEIGHT / 2` part already sets it to use 2 pages (each have the screen).
To now draw on the entire screen, you have to run the draw code twice. and disable the full screen update.
For this you have to replace this:
```c++
        DisplayWeather();
        display.display(false); // Full screen update mode
```
with this:
```c++
        display.firstPage();
        do {
          DisplayWeather();
        } while (display.nextPage());
```

CargoBikoMeter, 2021-08-30:
Hi Georg

thanks, now it works perfectly and I can combine the weather display with my Telraam traffic data display and the OpenSensMap PM sensor data display, which are currently under development. Yesterday I had successfully parsed and summarized the huge Telraam response data (https://documenter.getpostman.com/view/8210376/TWDRqyaV#3bb3c6bd-ea23-4329-b885-0d142403ecbb).

Now I have to look for a nice case for the 7.5" ePaper display, to mount the display near our Telraam sensor in the window. 

Roland