G6EJD/ESP32-e-Paper-Weather-Display #157: 4.2 inch version not displaying anything

EagleeyeKai, 2021-09-03:
Hi,
I already used this lovely peace of software but it is not working any longer.
I am using several different ususal §SP32 dev boards and the Waveshare 4.3 inch EPD. In the arduino IDE, I updated all libraries and board definitions. The EPD is connected as described and my own credentials are set and checked.

However, after compiling the sketch and uploading it to the board, the display stays white / unchanged. The serial monitor shows that a connection to my WifI was established but nothing else happens.

Is there a known problem with the newest versions of the libraries? Is there any chance to debug my problem?

Any help is highly appreciated.

Thanks!!

PS: A hello world Test sketch works perfectly...

AWSW2017, 2021-09-03:
Hi, 
I had something similar - as it sounds to me from your description…

Is the board able to reach the named time server ?

HTH

EagleeyeKai, 2021-09-03:
How can I check this?

G6EJD, 2021-09-03:
In the credentials file is a test url, enter your API key there and then click on the line it should return the JSON data

EagleeyeKai, 2021-09-03:
The test link is regarding OWM, isnt it? But this works and shows the weather data... I can also ping the time server 0.europe.pool.ntp.org from my terminal

G6EJD, 2021-09-03:
Which ESP32 extension are you using 1.0.6?
IDE 1.8.13?
Are you using a known good city location? Try London or Paris
The serial port should show the weather data being received it’s doing no more than the browser is doing.

EagleeyeKai, 2021-09-03:
IDE 1.8.13 and ESP32 V1.0.6

I was using Frankfurt, germany, but tried London, UK 

`
String City             = "London";                      // Your home city See: http://bulk.openweathermap.org/sample/
String Country          = "UK";                            // Your _ISO-3166-1_two-letter_country_code country code, on OWM find your nearest city and the country code is displayed
                                                           // https://en.wikipedia.org/wiki/List_of_ISO_3166_country_codes
String Language         = "EN";                            // NOTE: Only the weather description is translated by OWM
                                                           // Examples: Arabic (AR) Czech (CZ) English (EN) Greek (EL) Persian(Farsi) (FA) Galician (GL) Hungarian (HU) Japanese (JA)
                                                           // Korean (KR) Latvian (LA) Lithuanian (LT) Macedonian (MK) Slovak (SK) Slovenian (SL) Vietnamese (VI)
String Hemisphere       = "north";                         // or "south"  
String Units            = "M";                             // Use 'M' for Metric or I for Imperial 
//const char* Timezone    = "GMT0BST,M3.5.0/01,M10.5.0/02";  // Choose your time zone from: https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv 
                                                           // See below for examples
const char* ntpServer   = "0.europe.pool.ntp.org";                  // Or, choose a time server close to you, but in most cases it's best to use pool.ntp.org to find an NTP server
                                                           // then the NTP system decides e.g. 0.pool.ntp.org, 1.pool.ntp.org as the NTP syem tries to find  the closest available servers
                                                           // EU "0.europe.pool.ntp.org"
                                                           // US "0.north-america.pool.ntp.org"
                                                           // See: https://www.ntppool.org/en/                                                           
int   gmtOffset_sec     = 0;    // UK normal time is GMT, so GMT Offset is 0, for US (-5Hrs) is typically -18000, AU is typically (+8hrs) 28800
int  daylightOffset_sec = 3600; // In the UK DST is +1hr or 3600-secs, other countries may use 2hrs 7200 or 30-mins 1800 or 5.5hrs 19800 Ahead of GMT use + offset behind - offset

// Example time zones
//const char* Timezone = "MET-1METDST,M3.5.0/01,M10.5.0/02"; // Most of Europe
const char* Timezone = "CET-1CEST,M3.5.0,M10.5.0/3";       // Central Europe
`

After compiling and uploading, the serial monitor shows the following:

17:34:21.160 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:34:21.160 -> configsip: 0, SPIWP:0xee
17:34:21.160 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:34:21.160 -> mode:DIO, clock div:1
17:34:21.160 -> load:0x3fff0018,len:4
17:34:21.160 -> load:0x3fff001c,len:1216
17:34:21.160 -> ho 0 tail 12 room 4
17:34:21.194 -> load:0x40078000,len:10944
17:34:21.194 -> load:0x40080400,len:6388
17:34:21.194 -> entry 0x400806b4
17:34:21.473 -> 
17:34:21.473 -> Connecting to: Schwalbennest
17:34:23.730 -> WiFi connected at: 192.168.0.41
17:34:23.835 ->                                                                                                                                                                                                                                                                                                                                                                            

G6EJD, 2021-09-03:
can you send me the code to david.g6ejd@gmail.com

EagleeyeKai, 2021-09-04:
I just sent you the code, the compiler and terminal output..

EagleeyeKai, 2021-09-05:
David, thanks a lot for helping me out and providing the files that helped me compile the project. Now it works :)

closed 2021-09-05 by EagleeyeKai

G6EJD, 2021-09-05:
Good to hear, it was unfortunately the ArduinoJSON library that was causing the problems, I have already updated the library here, but will refresh all the files when I’ve tested each one.