G6EJD/ESP32-e-Paper-Weather-Display #158: HttpClient not found

MinitechnicusM, 2021-09-11:
Hi,
I try to compile your sample for e-Paper 7.5 and I got this error:
In file included from C:\Users\<user>\AppData\Local\Temp\arduino_modified_sketch_518961\Waveshare_7_5.ino:93:

C:\Users\<user>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP32-e-Paper-Weather-Display\src/common.h:6:10: fatal error: HTTPClient.h: No such file or directory

    6 | #include <HTTPClient.h>

      |          ^~~~~~~~~~~~~~

compilation terminated.
The file is present here:
C:\Users\<user>\OneDrive\Documents\Arduino\libraries\HttpClient

Any advice on this?
Thank you!

G6EJD, 2021-09-11:
Httpclient is a default library for the ESP32 compiler, so you have it installed, go to boards, manager and search for esp32 you should have 1.0.6 installed

MinitechnicusM, 2021-09-11:
My bad. I have ESP8266 and I didn't pay attention this is an ESP32 project. Thank you!

closed 2021-09-11 by MinitechnicusM