G6EJD/ESP32-e-Paper-Weather-Display #161: Waveshare 4.2 3C not displaying

BTMC1, 2021-10-04:
Hello , 

First thanks for your program , 

I'm encountering a problem with a new waveshare 4.2 3C I bought  . 

I switched the line corresponding to the new model GxEPD2_3C<GxEPD2_420c, GxEPD2_420c::HEIGHT> display(GxEPD2_420c(/*CS=D8*/ EPD_CS, /*DC=D3*/ EPD_DC, /*RST=D4*/ EPD_RST, /*BUSY=D2*/ EPD_BUSY));

Here are my pins for my firebeetle V4.0 : 
static const uint8_t EPD_BUSY = 27;  // to EPD BUSY
static const uint8_t EPD_CS   = 9;  // to EPD CS
static const uint8_t EPD_RST  = 16; // to EPD RST
static const uint8_t EPD_DC   = 17; // to EPD DC
static const uint8_t EPD_SCK  = 18; // to EPD CLK
static const uint8_t EPD_MISO = 19; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 23; // to EPD DIN

The program is running well and i'm able to have all data on series monitor , but nothing is displaying on the screen , I don't know where to look , here is the end of the message each reset : 

_PowerOn : 5
_Update_Full : 1
_PowerOff : 1
_PowerOff : 1

Triple checked cables and pins . Let me know if you have a solution to my problem . Thanks & have a great day !

G6EJD, 2021-10-04:
static const uint8_t EPD_CS = 9; // to EPD CS

Pin-9 is already in-use for the on-board SPI flash SD2, choose another spare pin and update your pin definitions to match

closed 2021-10-04 by G6EJD

BTMC1, 2021-10-05:
Wow , problem solved , thanks a lot !!! You saved me a lot of time :)