G6EJD/ESP32-e-Paper-Weather-Display #162: Waveshare 4.2, Column goes to grey scale

muelli01, 2021-10-07:
I have a strange issue with a Waveshare 4.2 display module.
This is the module I use:
https://www.waveshare.com/product/displays/e-paper/epaper-2/4.2inch-e-paper-module.htm

One column seems to be displayed in gray scale tone, others are black as they should be.
Looks like this (highlighted the column):
https://postimg.cc/cKW0ZtNp

First I thought the display or controller/driver was faulty, but when I tried a simple "hello world" programm or displayed some bitmap, everything showed black as the other columns.
Code on display init seems fine here, so I am quite puzzled on how to fix this.
My only idea is, that there is something not 100% correct in the refresh code for this certain display. Might not even be an issue for this repo, but more for gxEPD2?
Would appreciate any hint or advice.

G6EJD, 2021-10-07:
OK, it’s puzzling and because I have a test setup I’ve just tried the code and it works as expected.
Your using this line in display_init:
display.init(115200, true, 2, false);

What is the serial port log saying towards the end, is it taking just 3-4 secs in total or longer, for example if you watch it in real time does it have a pronounced delay before entering sleep.

muelli01, 2021-10-07:
>>Your using this line in display_init:
>>display.init(115200, true, 2, false);

Exactly

Serial output from the end:

11:01:21.295 -> _PowerOn : 39001
11:01:22.985 -> _Update_Full : 1517001
11:01:23.051 -> _PowerOff : 19001
11:01:23.051 -> _PowerOff : 0
11:01:23.051 -> Entering 220-secs of sleep time
11:01:23.051 -> Awake for : 5.388-secs
11:01:23.051 -> Starting deep-sleep period...

I use this ESP32 board:
https://www.az-delivery.de/products/esp32-developmentboard
It is nothing special.

G6EJD, 2021-10-07:
OK, nothing strange there then. Unfortunately my display board has no version number on it, it's about 3-months old, but works ok. Try this:#

At line 127: display.display(false); // Full screen update mode

Change it to:
 
 display.firstPage();
 do  { } while (display.nextPage());

muelli01, 2021-10-07:
Page buffer shows the same result.

G6EJD, 2021-10-07:
OK, try a delay after:
display.firstPage();
do { } while (display.nextPage());
delay(500);

It could be the display is slower now and needs times to complete the display before the ESP goes to sleep.

I notice in the specification that a full refresh takes 4-secs, and yet mine takes ~1-sec, the actual specification was 1.2-secs

G6EJD, 2021-10-07:
What display object line have you used?

The delay makes sense because if the display has not completed its update and the cpu goes to sleep before the display completes then it can leave a partial image on the display, giving it time enables it to complete, but it seems it's not that.
You can call DisplayWeather() multiples times during the page refresh, but I don't think that is the issue, it is something to do with your new display, but what?

G6EJD, 2021-10-07:
I've just checked the GxEPD2 non-paged update example and its identical, to my implementation, so this is not a software issue.

These problems are almost always to do with the Busy line, what are your pin definitions?

muelli01, 2021-10-07:
>What display object line have you used?

// Connections for e.g. LOLIN D32
static const uint8_t EPD_BUSY = 4;  // to EPD BUSY
static const uint8_t EPD_CS   = 5;  // to EPD CS
static const uint8_t EPD_RST  = 16; // to EPD RST
static const uint8_t EPD_DC   = 17; // to EPD DC
static const uint8_t EPD_SCK  = 18; // to EPD CLK
static const uint8_t EPD_MISO = 19; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 23; // to EPD DIN

GxEPD2_BW<GxEPD2_420, GxEPD2_420::HEIGHT> display(GxEPD2_420(/*CS=D8*/ EPD_CS, /*DC=D3*/ EPD_DC, /*RST=D4*/ EPD_RST, /*BUSY=D2*/ EPD_BUSY));

Ah OK, got the sense of the delay now. Just preventing an immediate sleep... Display update is reasonably fast around 1s i guess. So it seems it is in the range of your sample.

>You can call DisplayWeather() multiples times during the page refresh, but I don't think that is the issue, it is something to do with your new display, but what?

Calling it multiple times makes no difference in theory, as this just fills the pageBuffer (and does not in reallity, I checked).
It must have something to do with the display itself, but then again...why are the examples working?
Weird.

muelli01, 2021-10-07:
Got some interesting results on these examples.

First, they both basically work and stuff is shown on the screen. BUT....
...when an new image is shown, I can see faint shadows of what was before on this screen and the image is slightly grey-ish.
Right up until the moment when the time runs out and a new image should be shown.
Then, for the last few moments, the image turns to really dark black on those places, where the new image will be show.
Even, if those parts of the new image are white, they still turn the old image black at those places.
Hard to explain. Check attached foto for what I mean about shadows etc

[image]

G6EJD, 2021-10-07:
And with the weather display?

G6EJD, 2021-10-08:
So the code  is working then, if the panel is displaying grey or streaks then it’s faulty you need to contact the supplier.

closed 2021-10-08 by G6EJD

muelli01, 2021-10-08:
Indeed, they got back to me with some generic code from
https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board
that starts a webserver, lets you select the display and upload an grey/black image and displays it. the complete line goes to grey. It is cleary not a code problem or library problem.

Thanks for your effort!