G6EJD/ESP32-e-Paper-Weather-Display #171: problem with waveshare 4.2 e-Paper Module with 3 Colors , Version V2

xsnoopy1958, 2022-01-23:
Hi, 
I am a beginner and need help. Have sent the code to my ESP32 and display is connected, but there is nothing to see on the display.
IS there a problem with Version V2 of the 4.2 Display?

Could you please elp?

G6EJD, 2022-01-23:
There are no problems with the code, most likely incorrect connections or a cable fault if using DuPont cables.
Which ESP32 board type and display?

xsnoopy1958, 2022-01-23:
Hi, 
Thank you very much for your answer. 

I would very much like to get the weather station working.
But I am beginner and do not know about ESP32 and epaper Display. 

Display:  waveshare 400x300, 4.2 inch E-Ink Display Module, three-color

 ESP-Board: AZ Delivery ESP32 devkit C V2 with ESP32-WROOM 3

Could you please help? 

Thank you very much

Bernd

Sent from MailDroid

https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/issues/171#issuecomment-1019539717

G6EJD, 2022-01-23:
When you open the IDE monitor from the boards menu and reset the ESP so you see the weather data?

xsnoopy1958, 2022-01-23:
I will try it tomorrow morning. 

Sent from MailDroid

https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/issues/171#issuecomment-1019552311

xsnoopy1958, 2022-01-28:
now I have changed from ESP32 dev (from AZ delivery) to LOLIN32 and with this one it works.
I used the display entry: GxEPD2_3C<GxEPD2_420c (because of my 3 color display).

now I have some additional quesstions:
1) is the flickering during data update normal (see video)?
https://user-images.githubusercontent.com/58209306/151526315-9e15b4de-5ccb-42c8-8127-45a5c8d5d610.mp4

2) would it be possible to change the wifi data and weather data depending of the actual geo position during traveling?

3) Is it possible to integrate data from external temperature sensor?

G6EJD, 2022-01-28:
If you have no experience of e-paper displays then the flickering as it clears all 3 colours can be disconcerting, but it’s all normal and the 3-colour displays can take up to 12-secs to be cleared. A b/w variant takes just 1-sec.

Code changes, yes just about anything is possible and adding a local sensor would be easily achieved with a function to read the sensor and for ease of use assign the results to some global variables assigned to the sensor these can then be display on the screen using the display text function.

Yes the code could be modified to accept an input of location either by name and country of Lat/Lon and then get data for that location, the best method to do that would be a webserver.

the code would need considerable change for the location change, the sensor just a few lines.

closed 2022-01-28 by G6EJD