G6EJD/ESP32-e-Paper-Weather-Display #172: region `dram0_0_seg' overflowed

miszowaty77, 2022-01-24:
Hi,

trying to load code into TTGO Lora32 v2.1 and size of the program seems to be too big for memory. 

`…/xtensa-esp32-elf/bin/ld: Waveshare_7_5_T7.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
.../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
…/xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 17480 bytes
collect2: error: ld returned 1 exit status`

I am thinking of assigning more memory. What's the way to handle it?

G6EJD, 2022-01-24:
Try using vroom board type and reduce the SPIFFS size down, 3MB programme and 1MB spiffs in the board options on the same menu

Regards

David

miszowaty77, 2022-01-24:
Can't see this option in menu: [](https://imgur.com/a/rXj7sSB)

G6EJD, 2022-01-24:
Did you select Vroom board type?

miszowaty77, 2022-01-24:
Oh that's right, the menu is now expanded. I choose "huge APP" but still same error.

miszowaty77, 2022-01-24:
Do you mean PSRAM? Already enabled. I am using BWR display, maybe driver with 3 colors is too big?

G6EJD, 2022-01-24:
Well that’s the reason, there’s no way an ESP32 can handle the memory requirements for a 3-colour display unless the code is radically revised

closed 2022-01-24 by G6EJD

miszowaty77, 2022-01-24:
Do u think that I can drive 3 colors display with BW driver?

G6EJD, 2022-01-24:
No, only the B/W driver will work.