G6EJD/ESP32-e-Paper-Weather-Display #174: missing battery voltage display

xsnoopy1958, 2022-02-21:
Hi,
I am using your ESP32-e-Paper-Wather-Display with waveshare 4,2 inch display and LOLIN32.
It  is very impressive and i like it very much.
But I miss the battery voltage display on the e-Paper.

Can you help please?
Thank you in advance.

Bernd

G6EJD, 2022-02-21:
That’s because your board is a copy Lolin32 and does not include the voltage sensing input. To fix this you’d need to wire 2 resistors as a voltage divider, so one resistor connects to the Vbat or Battery + terminal the other end to GPIO 35 then from GPIO 35 to Ground via another resistor, both values should be equal and of value 330K each.

closed 2022-02-21 by G6EJD

G6EJD, 2022-02-23:
Good to hear that’s working.

Day-0 is today and then so-on, there was not enough space on screen to put a full word. You  can get the time if you wish from the JSON data, the functions are in the code to do that.