G6EJD/ESP32-e-Paper-Weather-Display #175: Waveshare 2.66 (Red/Black) and  5.83 (B/W) working??

g0emf, 2022-02-21:
Hello, just a quick question
I have 
https://www.waveshare.com/2.66inch-e-Paper-Module-B.htm
and
https://www.waveshare.com/5.83inch-e-Paper-HAT.htm (display and hat)

My question is did anyone get these to work ?

Got the for Xmas and found time to have a play, I have spent many days or weeks trying to get them up and running.
Been through the internet many times and have not found a working solution.

I won't go through all the configs and fixes I have used, I only ask if the above models are working with a ESP32 
https://www.ebay.co.uk/itm/322484359954

Thanks for any information guys
Just I don't want to spend more time looking for a configuration/library that might not exist at this date.

G6EJD, 2022-02-21:
The 5.83” display is certainly supported and works well. The 2.66” version looks remarkably like the 2.9” variant and has similar resolution so I would try that driver object.
That ESP32 board is very expensive and will consume a lot of power and it not recommended, the genuine Lolin D32 is the best board to get and much lower cost.

closed 2022-02-21 by G6EJD

g0emf, 2022-02-21:
Hi & thanks for the info, I did get the 2.66 working with the 2.9 variant (as expected the top bit of the screen had random bits in it)
However, I have not had anything on the 5.83.
I will spend another day or so on it but after your reply, I suspect I might have a defective 5.83 screen?
Thanks for the info de Dave g0emf
73

G6EJD, 2022-02-21:
I’ve got a 5.83” display and just plugged it in, works without issues, so yes suspect yours is faulty. One way of testing is to get the 2.66@ display working then unplug and fit the larger display then that reduces the chances of a connection error, the plug is identical for all screen sizes, obviously need to reprogram the ESP32, 73

g0emf, 2022-02-21:
Yup did that last week, still nothing on the display.Thanks73