G6EJD/ESP32-e-Paper-Weather-Display #178: Trying to compile the WAVESHARE_7_5_T7 sketch on a fresh Arduino IDE install fails

devoid1, 2022-03-05:
followed the directions.  other people seem to be using the sketch fine but im not sure if they all have other models of the screen.  i have the v2 waveshare 7.5" e-paper.  i load arduino ide, add required libraries, and it wont compile due to missing HTTPClient.h.  all the http client libraries i can find are spelled differently HttpClient.  when i change the sketch and the obtain_MX function (in common.h) to the other spelling it chokes on other syntax errors.  apparently i dont have the correct library somewhere.  someone on reddit suggested that perhaps the ESP32 library has now been superseded by the ESP8266 library.  could that be a problem?  i just want the thing to compile.

arduino ide 1.8.19
when i change the HTTPClient to HttpClient in common.h i get:
    Multiple libraries were found for "HttpClient.h"

    Used: /home/darren/Arduino/libraries/ArduinoHttpClient

    Not used: /home/darren/Downloads/arduino-1.8.19/libraries/Bridge

    exit status 1

    no matching function for call to 'HttpClient::HttpClient(WiFiClient&)'

G6EJD, 2022-03-05:
Have you installed the ESP32 compiler?
Version?

G6EJD, 2022-03-05:
I've just compiled this version and no errors found.

You've most probably installed the incorrect HTTP library. There is no need for an external library as the correct one is in-built with the compiler. Only those libraries that I reference at the start are required from external sources.

closed 2022-03-05 by G6EJD