G6EJD/ESP32-e-Paper-Weather-Display #189: Problem using Waveshare e-Paper ESP32 Driver Board.

RomanFoltyn, 2022-12-29:
TEST - Program version for "Waveshare_7_5_T7" display version 16.11

After starting the program, the display shows the content of the weather station image in the first second in negative (black background), then a white screen, then inverted again, and finally a white screen with almost unrecognizable content (very light gray).

Exactly the same program using the ESP32 Dev module + e-Paper Driver HAT works great. Finally, after the last image in the negative flashes, a wonderful high-contrast image appears. Of course, the Pins for both ESPs have been selected accordingly.

I thought my two e-Paper ESP32 Driver Boards were damaged but it turns out that the Demonstration program for this display provided with Waveshare "epd7in5_V2-demo" works fine.

I checked the diagrams matching ESP to 24 pin and they are practically the same with the exception of e-Paper Driver HAT reset (LOW) of the display turns off the power of the Display.
However, checking this voltage with an oscillograph, I found that the Reset is so short that the voltage remains because the capacitors support it at this point.

G6EJD, 2022-12-29:
It’s a GxEPD2 issue, he provides a facility for extending reset times and there are examples on his GitHub. The parameter is set in the .init(x,y,z); function

closed 2022-12-29 by G6EJD

RomanFoltyn, 2022-12-29:
**Thank you for your answer**, but I have a question because I checked and I have init(115200, true, 2, false) in the program a long time ago.
there is still 1 kOhm left for Reset ESP32. ESP32 only has Pin EN and it has PulUp1 K by default.
I guess it's about the RES pin of the display (Pin 10 of the ribbon)?
is that it?
Regards
Roman **DL6NBF** (ex SP9LDM)

G6EJD, 2022-12-29:
2 is 2mS it might need to be longer, there are problems with the reset circuit of some Waveshare HAT’s which is why the option was added. In the GxEPD2 readme there is a recent addition that explains the problems with some displays. Yes it’s all about the RESET line try a much longer reset time.

RomanFoltyn, 2022-12-29:
Hello David,

Unfortunately I'm desperate, I added a 1kOhm resistor to the ESP32 output 26 to the 3.3V E-Paper power supply (this is Pin 10 of the ribbon) but it did nothing.

for the command display.init(115200, true, 20, false); I tested values from 2 to 200 and even 1000 but all this still doesn't work properly.

 

Maybe you have ideas that don't fit yet,

Of course, I can use a "normal" esp32 module, but then I have to add an "e-paper driver hat" in which I would have to desolder the Raspberry sockets and the ESP 32 cable because they thicken the board.

But this solution, having already purchased three pieces of e-Paper ESP32 Board, is very inelegant!

 

I am asking you for help.

Regards Roman DL6NBF

G6EJD, 2022-12-29:
Hi Roman, clearly this is a hardware issue, but I’m drawn to the pin assignments of the Waveshare driver board which are quite different to the ESP32 Dev board, which does work. As the ESP32 itself is identical it has to be your pin assignments which pins are you using and are you trying to override the assignments made by the compiler, try printing out the SPI bus assignments of MISO, MOSI, SCK, DC, CS and RESET and make sure those results should match the Waveshare wiring.

RomanFoltyn, 2022-12-29:
Hello David,

 

I checked everything and the pins are set correctly.

The proof of this is that the display at startup shows a beautiful image of the weather station in negative a few times and then switches to white with a very light gray image (barely visible).

I thought that the Display hardware (power supply and capacitors are wrong, but if the Waveshare demo displays correctly, it can't be done wrong. 

 

As I wrote to you, the only difference in the diagrams is that the board turns off the panel power when reset. It is such a short time that the capacitor on the power supply maintains this voltage drop.

 

Roman  PS you better write to ***@***.*** because the other one gets too many spams

G6EJD, 2022-12-29:
Hi Roman, I don't have your email address all communication is via GitHub, so if your suffering from spam, it's best to update your GitHub account.
I have both Waveshare boards and a T7 display, please send me links to the exact boards your using and display so I can compare and check the schematics and also which GxEPD2 display constructor line your using near to the start of the programme. 

Dave

G6EJD, 2022-12-29:
Also, try adding a 2-sec delay before the sleep line is called so delay(2000); to allow the display time to settle, I notice in the latest Waveshare examples they are adding a 2-sec delay and say it's very important.

G6EJD, 2022-12-29:
So you should have:
// Connections for e.g. LOLIN D32
//static const uint8_t EPD_BUSY = 4;  // to EPD BUSY
//static const uint8_t EPD_CS   = 5;  // to EPD CS
//static const uint8_t EPD_RST  = 16; // to EPD RST
//static const uint8_t EPD_DC   = 17; // to EPD DC
//static const uint8_t EPD_SCK  = 18; // to EPD CLK
//static const uint8_t EPD_MISO = 19; // Master-In Slave-Out not used, as no data from display
//static const uint8_t EPD_MOSI = 23; // to EPD DIN

// Connections for e.g. Waveshare ESP32 e-Paper Driver Board
static const uint8_t EPD_BUSY = 25;
static const uint8_t EPD_CS   = 15;
static const uint8_t EPD_RST  = 26; 
static const uint8_t EPD_DC   = 27; 
static const uint8_t EPD_SCK  = 13;
static const uint8_t EPD_MISO = 12; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 14;

GxEPD2_BW<GxEPD2_750_T7, GxEPD2_750_T7::HEIGHT> display(GxEPD2_750_T7(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY));   // B/W display

G6EJD, 2022-12-29:
Do you have the driver board with two switches for the T7 switch A should be ON