G6EJD/ESP32-e-Paper-Weather-Display #190: How to connect e-paper?

dippaolo, 2023-01-15:
How to connect e-paper?

I have ESP-C3-32S.

Link foto: [https://photos.app.goo.gl/cQFq4HCnDeiVBdf9A](https://photos.app.goo.gl/cQFq4HCnDeiVBdf9A)

G6EJD, 2023-01-15:
Wire the pins using this diagram:RegardsDavid

dippaolo, 2023-01-15:
According to what scheme?
I have 15 pins.
I have other pins.

G6EJD, 2023-01-15:
The code sets the pins to use, so you have to select them on the hardware and wire the display accordingly. If you can’t select the board type in the IDE then you’ll find it very difficult this is what happens when you choose a non standard board, it needs work to determine what pins are available!

dippaolo, 2023-01-21:
OK.
Just tell me if my ESP is working properly. The serial port monitor shows:

17:45:12.030 -> ESP-ROM:esp32c3-api1-20210207
17:45:12.030 -> Build:Feb  7 2021
17:45:12.030 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
17:45:12.030 -> SPIWP:0xee
17:45:12.030 -> mode:DIO, clock div:1
17:45:12.030 -> load:0x3fcd5810,len:0x438
17:45:12.030 -> load:0x403cc710,len:0x918
17:45:12.030 -> load:0x403ce710,len:0x24e4
17:45:12.030 -> entry 0x403cc710
17:45:12.215 -> 
17:45:12.215 -> Connecting to: Paolo_
17:45:12.405 -> WiFi connected at: 192.168.10.122
17:45:13.997 -> Sat Jan 21 2023   16:45:17
17:45:13.997 -> Initialising Display
17:45:13.997 -> Begin InitialiseDisplay...
17:45:14.935 -> ESP-ROM:esp32c3-api1-20210207
17:45:14.935 -> Build:Feb  7 2021
17:45:14.935 -> rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
17:45:14.935 -> Saved PC:0x40380082
17:45:14.935 -> SPIWP:0xee
17:45:14.935 -> mode:DIO, clock div:1
17:45:14.935 -> load:0x3fcd5810,len:0x438
17:45:14.935 -> load:0x403cc710,len:0x918
17:45:14.935 -> load:0x403ce710,len:0x24e4
17:45:14.935 -> entry 0x403cc710
17:45:15.123 -> 
17:45:15.123 -> Connecting to: Paolo_
17:45:15.263 -> WiFi connected at: 192.168.10.122
17:45:15.355 -> Sat Jan 21 2023   16:45:18
17:45:15.355 -> Initialising Display
17:45:15.355 -> Begin InitialiseDisplay...
17:45:16.292 -> ESP-ROM:esp32c3-api1-20210207
17:45:16.292 -> Build:Feb  7 2021
17:45:16.292 -> rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
17:45:16.292 -> Saved PC:0x40380082
17:45:16.292 -> SPIWP:0xee
17:45:16.292 -> mode:DIO, clock div:1
17:45:16.292 -> load:0x3fcd5810,len:0x438
17:45:16.292 -> load:0x403cc710,len:0x918
17:45:16.292 -> load:0x403ce710,len:0x24e4
17:45:16.292 -> entry 0x403cc710
17:45:16.481 -> 
17:45:16.481 -> Connecting to: Paolo_
17:45:16.668 -> WiFi connected at: 192.168.10.122
17:45:16.761 -> Sat Jan 21 2023   16:45:20
17:45:16.761 -> Initialising Display
17:45:16.761 -> Begin InitialiseDisplay...
17:45:17.699 -> ESP-ROM:esp32c3-api1-20210207
17:45:17.699 -> Build:Feb  7 2021
17:45:17.699 -> rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
17:45:17.699 -> Saved PC:0x40380082
17:45:17.699 -> SPIWP:0xee
17:45:17.699 -> mode:DIO, clock div:1
17:45:17.747 -> load:0x3fcd5810,len:0x438
17:45:17.747 -> load:0x403cc710,len:0x918
17:45:17.747 -> load:0x403ce710,len:0x24e4
17:45:17.747 -> entry 0x403cc710
17:45:17.887 -> 
17:45:17.887 -> Connecting to: Paolo_
17:45:18.026 -> WiFi connected at: 192.168.10.122
17:45:18.119 -> Sat Jan 21 2023   16:45:21
17:45:18.119 -> Initialising Display
17:45:18.119 -> Begin InitialiseDisplay...
17:45:19.053 -> ESP-ROM:esp32c3-api1-20210207
17:45:19.053 -> Build:Feb  7 2021
17:45:19.099 -> rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
17:45:19.099 -> Saved PC:0x40380082
17:45:19.099 -> SPIWP:0xee
17:45:19.099 -> mode:DIO, clock div:1
17:45:19.099 -> load:0x3fcd5810,len:0x438
17:45:19.099 -> load:0x403cc710,len:0x918
17:45:19.099 -> load:0x403ce710,len:0x24e4
17:45:19.099 -> entry 0x403cc710
17:45:19.239 -> 
17:45:19.239 -> Connecting to: Paolo_
17:45:19.381 -> WiFi connected at: 192.168.10.122
17:45:19.521 -> Sat Jan 21 2023   16:45:22
17:45:19.521 -> Initialising Display
17:45:19.521 -> Begin InitialiseDisplay...
17:45:20.460 -> ESP-ROM:esp32c3-api1-20210207
17:45:20.460 -> Build:Feb  7 2021
17:45:20.460 -> rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
17:45:20.460 -> Saved PC:0x40380082
17:45:20.460 -> SPIWP:0xee
17:45:20.460 -> mode:DIO, clock div:1
17:45:20.460 -> load:0x3fcd5810,len:0x438
17:45:20.460 -> load:0x403cc710,len:0x918
17:45:20.460 -> load:0x403ce710,len:0x24e4
17:45:20.460 -> entry 0x403cc710
17:45:20.601 -> 
17:45:20.601 -> Connecting to: Paolo_
17:45:21.205 -> WiFi connected at: 192.168.10.122
17:45:21.298 -> Sat Jan 21 2023   16:45:24
17:45:21.298 -> Initialising Display
17:45:21.298 -> Begin InitialiseDisplay...
17:45:22.234 -> ESP-ROM:esp32c3-api1-20210207
17:45:22.234 -> Build:Feb  7 2021
17:45:22.234 -> rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
17:45:22.281 -> Saved PC:0x40380082
17:45:22.281 -> SPIWP:0xee
17:45:22.281 -> mode:DIO, clock div:1
17:45:22.281 -> load:0x3fcd5810,len:0x438
17:45:22.281 -> load:0x403cc710,len:0x918
17:45:22.281 -> load:0x403ce710,len:0x24e4
17:45:22.281 -> entry 0x403cc710

G6EJD, 2023-01-21:
Your ESP is continuously resetting and restarting, most likely a poor power supply or an incorrect connection.

dippaolo, 2023-01-25:
ESP32 damaged.
I bought a second one. It doesn't reset.

closed 2023-01-25 by dippaolo