G6EJD/ESP32-e-Paper-Weather-Display #192: 2.9'' V2 - unable to compile

jjvelar, 2023-02-17:
Hi,
I am not able to compile for 2.9'' display.
If I try the 2.7'' version, it compiles and uploads like a charm but does not show anything in the screen despite I have changed dimensions.
The problem with 2.9'' is even worst as I can't see it Examples, only the the 2_9_T5 version.
Any advise? Is there any additional information I could provide you?
Thanks and best regards

G6EJD, 2023-02-17:
What is the compiler error message?

G6EJD, 2023-02-17:
There are two 2.9" versions, which one are you using?

G6EJD, 2023-02-17:
You could have any of these display types:
//GxEPD2_BW<GxEPD2_290, GxEPD2_290::HEIGHT> display(GxEPD2_290(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEH029A1 128x296, SSD1608 (IL3820)
//GxEPD2_BW<GxEPD2_290_T5, GxEPD2_290_T5::HEIGHT> display(GxEPD2_290_T5(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEW029T5 128x296, UC8151 (IL0373)
//GxEPD2_BW<GxEPD2_290_T5D, GxEPD2_290_T5D::HEIGHT> display(GxEPD2_290_T5D(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEW029T5D 128x296, UC8151D
//GxEPD2_BW<GxEPD2_290_I6FD, GxEPD2_290_I6FD::HEIGHT> display(GxEPD2_290_I6FD(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEW029I6FD 128x296, UC8151D
//GxEPD2_BW<GxEPD2_290_T94, GxEPD2_290_T94::HEIGHT> display(GxEPD2_290_T94(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEM029T94 128x296, SSD1680
//GxEPD2_BW<GxEPD2_290_T94_V2, GxEPD2_290_T94_V2::HEIGHT> display(GxEPD2_290_T94_V2(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEM029T94 128x296, SSD1680, Waveshare 2.9" V2 variant
//GxEPD2_BW<GxEPD2_290_BS, GxEPD2_290_BS::HEIGHT> display(GxEPD2_290_BS(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // DEPG0290BS 128x296, SSD1680
//GxEPD2_BW<GxEPD2_290_M06, GxEPD2_290_M06::HEIGHT> display(GxEPD2_290_M06(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEW029M06  128x296, UC8151D

You'd need to try them all unless you can see some marking on the display.

jjvelar, 2023-02-17:
> What is the compiler error message?

I just can't find my version when showing all examples. I have 2.9'' version and I can only see 2.9'' T5 example despite there is a folder for my version (Waveshare_2_9)

jjvelar, 2023-02-17:
My display version is 2.9" v2

jjvelar, 2023-02-17:
[image]

as you can see, despite there is the folder Waveshare_2_9, I can't find it in the examples menu.

G6EJD, 2023-02-17:
Your copy of the repository is out of date. Either download it again or just go to the GitHub page just above here and copy the file out of there.

G6EJD, 2023-02-17:
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/tree/master/examples/Waveshare_2_9

jjvelar, 2023-02-18:
I have done it but it is still the same: Waveshare_2_9 example does not appear despite the folder is available.
I see 2_9 is v7 while 2_9_T5 is v6.5 as it includes 3-day forecast.
Could this make the difference?
Could I try the v6.5 version for 2_9?

G6EJD, 2023-02-18:
Let’s find out which display your using: do you have a link to the sellers webpage?

jjvelar, 2023-02-18:
Bought at Amazon: https://www.amazon.es/dp/B071JFRV2S?ref=ppx_yo2ov_dt_b_product_details&th=1

Sticker in the board "V2" and "Rev2.1"

Marca	‎Waveshare
Fabricante	‎Waveshare
Modelo	‎2.13inch E-Ink HAT 250x122
Dimensiones del producto	‎3.51 x 2.01 x 0.2 cm; 18.14 gramos
Número de modelo del producto	‎2.13inch E-Ink HAT 250x122
Número de producto	‎e-paper module
Fabricante del procesador	‎VIA
Tamaño de la pantalla	‎2.9 Pulgadas

G6EJD, 2023-02-18:
That link says 2.13”? Or is it 2.9”?

jjvelar, 2023-02-18:
2.9”

G6EJD, 2023-02-18:
Regardless of the display are you seeing weather data on the serial monitor port?

jjvelar, 2023-02-18:
No.
On top of that, get an error about not being able to get time, so no meteo
info available

jjvelar, 2023-02-18:
I also get “param reqd error” log

G6EJD, 2023-02-18:
That tells me it’s not getting an internet connection, have you set the credentials for your WiFi network and OWM API key?

jjvelar, 2023-02-18:
It gets an IP address but get stuck getting time

G6EJD, 2023-02-18:
Have you set the time server address in the credentials file though as default it will use the UK time. Is your Router blocking access to some services like time?
If it can’t get time it will automatically time out and proceed to get the weather data and display the results.

jjvelar, 2023-02-18:
here you've got my credentials file.
log is:
psram error, connects to my wify, gets an IP, does not get time and then
goes into deep sleep, so no meteo data available

[image: image.png]

G6EJD, 2023-02-18:
You should not not get a PSRAM error, which board type are you selecting, choose a Wemos D32.
It sounds like your API key for OpenWeatherMap is not valid.

jjvelar, 2023-02-18:
my board is an esp32-wroom-32
here you've got my board configuration in arduino ide. Should I try wemos
d32?
will double check openweathermap API key.

[image: image.png]

G6EJD, 2023-02-18:
Yes try the default vroom board at the top of the list or the D32 type.
There is an example API call in the credential file put your API in that and try pasting it in your browser address bar it should return the weather

jjvelar, 2023-02-18:
will try to compile tomorrow with the other boards but, as you can see, API
key is working OK
anything else I could try?

[image: image.png]

G6EJD, 2023-02-18:
If the API key is working then you should get the same result on the serial port whether the display is connected or working or not, same for the time. 
Testing the API key with a browser proves the Router is not blocking anything, so it was the wrong board type.

jjvelar, 2023-02-18:
changed the board and now I get meteo data in serial monitor!
but screen does not update/change...
what else should I try?

G6EJD, 2023-02-18:
OK, screen aside it’s all working now, that’s good progress. Now you need to replace the line in the source code with each GxEPD2 driver object until a match with your display is found, see my post above with the examples that need trying one by one.  It might be the o d with V2 in the name. But keep a copy of the code you have running now so you can backtrack if it all goes wrong.

jjvelar, 2023-02-19:
That one did the magic:

GxEPD2_BW<GxEPD2_290_T94, GxEPD2_290_T94::HEIGHT> display(GxEPD2_290_T94(/*CS=D8*/ EPD_CS, /*DC=D3*/ EPD_DC, /*RST=D4*/ EPD_RST, /*BUSY=D2*/ EPD_BUSY)); // GDEM029T94 128x296, SSD1680

Thanks a lot for your support!

G6EJD, 2023-02-19:
Happy to help.

closed 2023-02-19 by G6EJD