G6EJD/ESP32-e-Paper-Weather-Display #193: 3-day forecast: the 3d day is not visible

jjvelar, 2023-02-19:
Hi,

Noticed there was an empty square in the 3d position of the 3-day forecast bottom view in my 2.9" display.
Solved by changing max_readings value from 25 to 33 in ino file but don't know if this the right way to solve it

#define max_readings 33 // In groups of 3-hours (3-days = 3 x 8 = 24)

jjvelar, 2023-02-19:
will report again if only 2-days

closed 2023-02-19 by jjvelar

G6EJD, 2023-02-19:
One of those codes was modified by someone else, check if the for next loop is doing 1 to 2 or 0 to 2, also check the times. Also OpenWeatherMap does not update the forecast until about 10:00 so it may not be broadcast early on but then appears.

jjvelar, 2023-02-19:
It is now working like a charm.
Quick question: can  the 2.9” show battery level?

G6EJD, 2023-02-19:
Some ESp32 boards have a on-board battery monitor, usually on pin 35, if not choose an ADC pin say 36 or 39 and wire two resisters to the board, then change the battery routine in the code to use the chosen pin.
Battery----R1----ADC Pin 36----R2----Gnd

Where R1 is about 200K and R2 100K, it will need calibrating. Just vary the multiplier on the readadc line.

jjvelar, 2023-02-19:
Way beyond and above me, but thanks!

jjvelar, 2023-02-20:
I hope you don't mind if I ask you another question: is a Wemos D1 Mini
enough for a 2.9" display? I am currently using an ESP32 but thought it
might be a little bit overkilling.
Thanks and best regards,
José

G6EJD, 2023-02-20:
Hello José, the 8266 has a lot less memory than the ESP32 needed to decode the json weather data and form the display image, I have never tried it, it may work, the code would need to be modified in a few areas to operate as some functions are not supported, David

jjvelar, 2023-02-20:
I will then keep the esp32
Thanks