G6EJD/ESP32-e-Paper-Weather-Display #197: Pressure trend - false direction

dreamy1, 2023-03-05:
Hello,

just found out that the trend for the pressure is displayed wrong:

    float pressure_trend = **WxForecast[0].Pressure - WxForecast[2].Pressure**; // Measure pressure slope between ~now and later
    pressure_trend = ((int)(pressure_trend * 10)) / 10.0; // Remove any small variations less than 0.1
    WxConditions[0].Trend = "0";
    if (pressure_trend > 0)  WxConditions[0].Trend = "+";
    if (pressure_trend < 0)  WxConditions[0].Trend = "-";
    if (pressure_trend == 0) WxConditions[0].Trend = "0";

In my opinion, the first line should be:
**float pressure_trend = **WxForecast[2].Pressure - WxForecast[0].Pressure**;

G6EJD, 2023-03-05:
If the pressure 2-hours ago was 1010 and now it is 1000 then the pressure is falling (1000-1010) and the symbol is ‘-‘

And your saying:

If the pressure 2-hours ago was 1010 and now it is 1000 then the pressure is ? (1010-1000) and the symbol is ‘+’

So which is right?

dreamy1, 2023-03-05:
Yes, but you have not the historical data. You compare the pressure at the moment (data 0) with the pressure in future (data 2).

So it is wrong. Take a look at your picture of the display, is shows "rising" but the pressure will be falling in the next hours: https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/master/Waveshare_7_5_new.jpg

G6EJD, 2023-03-05:
Which code are you referring to, your saying an image from another piece of code is wrong, and yet that code uses the OWM .trend value?

dreamy1, 2023-03-05:
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/master/examples/OWM_75_epaper_v16_7/OWM_75_epaper_v16_7.ino

The .trend value is not an OWM API value, it is generated inside the code here: https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/master/src/common.h

G6EJD, 2023-03-05:
Corrected

closed 2023-03-05 by G6EJD