G6EJD/ESP32-e-Paper-Weather-Display #201: Add probability of precipitation to timeseries graph?

jpolton, 2023-03-26:
Probability of precipitation is downloaded with the 3hrly forecast data. And is quite interesting.
This can be overlayed on the precipitation rate graph with a separate y-axis (on e.g. RHS of rainfall graph).
On a 7.5" screen this fits fine with a little bit of nudging of the other elements in the DisplayForecastSection().

[image]
[image]

Would this be of interest as a Pull Request?

dreamy1, 2023-03-27:
Yes :-)

jpolton, 2023-03-28:
Started a PR: [#202](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/pull/202).
Implemented in Waveshare_7_5_T7.ino only, for now.

jpolton, 2023-04-06:
Edits should work on other screens but the forecast graph spacing might have to be tweaked to get the extra y-axis labels to fit.

For the 7.5" 800x480 screen this was a simple edit in the DisplayForecastSection() method:
```
int gx = (SCREEN_WIDTH - gwidth * 4) / 5 + 5;
```
-->
```
int gx = (SCREEN_WIDTH - gwidth * 4) / 5 + 1;
```

I am reluctant to make a PR for things I can't personally test, unless someone else wants the PR and can help with that.

jpolton, 2023-05-06:
> Yes :-)

@dreamy1  Let me know what board you have, I'll make the changes for it and you could see if it looks OK?

dreamy1, 2023-05-06:
Thanks a lot - i have the 7.5" and the Liliygo T5 4.7" screen...but at the moment, unfortunately no time for testing...

Kobbe1, 2023-05-06:
I have a 4.2 inch board. I did not want to wait so already more or less implemented in my local copy which is heavily customized.
I really love this feature - been using it since a few weeks now, thanks for the awesome idea.
I did not actually add an Y-axis to show the % - as it is always 0 - 100% I don't think it is really needed and things are tight on a 4.2 screen.

I might be able to test it some time in the future when I merge other new changes but I don't regularly update it if all is well as it is "in production" - with the family, so changes are risky :)

G6EJD, 2023-05-06:
I agree, actually OWM’s predicted rainfall amounts to gives a profile of predicted precipitation without the need to draw it as yet another graph line.

jpolton, 2023-11-10:
Associated PR #202 got closed when I attempted to move it off my master branch. Created a new [PR](#232)

closed 2024-10-11 by G6EJD