G6EJD/ESP32-e-Paper-Weather-Display #204: Usage of "MostlySunny()" for OWM Icon "04n"/"04d" wrong?

Oposum01, 2023-04-15:
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/37afbd1a0faff0b20f76ede12214a2ff3f6e454d/examples/Waveshare_4_2/Waveshare_4_2.ino#L415

I'd like to discuss if the usage of "MostlySunny()" for the Icon "04n" / "04d" is wrong. According to https://openweathermap.org/weather-conditions I'd change it to "MostlyCloudy()", as well as removing the "addsun()" in that function too, which makes no sense to me there. "MostlySunny()" is already used for "02d" or "02n", which looks right to me.

G6EJD, 2023-04-16:
When I developed the code and looked at the OWM icons especially the three main indicators of Sunny, MostlyCloudy and Cloudy.
My logic was:
Sunny is a sun symbol.
Cloudy is two cloud symbols.
MostlyCloudy is not full cloud so must have interspersed sun, so I drew a cloud, then added a sun and then a cloud over the top, to denote there were bright areas, noting there was no colour to use other than black and white.

I’m not sure what else could be done.

also what’s in a name I called it MostlyCloudy and OWM called it ‘Few Clouds’ but the result is the same a cloud plus sun set of symbols?

Oposum01, 2023-04-16:
Thanks for your quick reply and explanations.
I've to admit, it depends on everyones own interpreation how to map the icons from OWM to the ones in your code, sure. 

My understanding of the OWM icons is, that it goes from "best" weather to "worst", see my lousy paint image:

[image]

Maybe I repeat myself, but it looks weird to me, that MostlySunny() is used for two icons, which represents different weather situations. I understand your explanation, but I still think that your interpretation is not appropriate for "broken clouds" (04d / 04n). It's obvious that sunlight could break through clouds here and there.

I stumbled across this, because we have a weather situation for several days, where it's cloudy all the time. During the day there is sunlight, but it will not break through the clouds so much. Your current code displays the MostlySunny() for that situation.
You've even introduced MostCloudy(), but it's never getting used in the code.

>also what’s in a name I called it MostlyCloudy and OWM called it ‘Few Clouds’ but the result is the same a cloud plus sun set of symbols?

I am not sure, if I got your question right, but I did not want to discuss about 02d/02n -> "few clouds" / MostlySunny().

G6EJD, 2023-04-17:
I've changed that now to MostlyCoudy and improved the wind direction calculation.

closed 2023-04-17 by G6EJD