G6EJD/ESP32-e-Paper-Weather-Display #205: Size of Weather Pictures

gopher333, 2023-04-16:
First let me thank you ffor this fantastic Program! It works instantly!
Can you pls tell me the exact size of the Pictures for the weather situation top right? I want to put in own Bitmaps
Tks in advance

G6EJD, 2023-04-16:
Top right icons? Not sure which you mean, please expand. The main icons are on a 100x100 grid, smaller types on a 50x50 grid.

On the basis that you went to conserve battery power to get maximum battery life then bitmap drawing takes a long time and so using them will reduce battery life.

gopher333, 2023-04-16:
I mean the Pictograms of the Weather situation (Person with umbrella and so on). They are in the file Micro_Wx_Icons.h
Already had success in putting in other graphics. Using 1,54 Waveshare Display
[image]

G6EJD, 2023-04-16:
Those are 128x128 bits, I was confused because they are in the weather girl code section

closed 2023-04-16 by G6EJD