G6EJD/ESP32-e-Paper-Weather-Display #207: Sun Icon During Nighttime

Jarod96, 2023-04-21:
I'm looking for a way to get rid of the sun icon during nighttime, say between sunset and sunrise.
is there a change that can be done easily without having to rewrite everything? how to proceed?
please enlighten me!
Thank you,
[image]
EPD 5.83inch

G6EJD, 2023-04-21:
The OWM icons are the same day and night, so I added the moon symbol. 
you could in each function add a test in from of the addsun function to only draw it when there is no moon request.

G6EJD, 2023-04-21:
if (!IconName.endsWith("n")) addsun(..

closed 2023-04-21 by G6EJD

Jarod96, 2023-04-21:
Thanks for your helpful reply, I was struggling to figure out a way to fix that!
What is the condition that make a moon request? I've searched everywhere in the code to understand how it work but without success...
Also, can I make the sun appear only after sunrise and before sunset? since sometime the moon is up there in the sky with the sun altogether

G6EJD, 2023-04-21:
OWM issue the icon number eg 01d or 01n hence the test checks endswith ‘n’  they calculate when it is passed sunset for your region and add ‘n’ at night