G6EJD/ESP32-e-Paper-Weather-Display #212: Waveshare 7.3inch e-Paper (F)

paul179, 2023-05-09:
Hello, 
I'm trying to get the e-paper display Waveshare 7.3inch e-Paper (F) to work, but unfortunately I can only see the lower quarter of the display.
[https://www.waveshare.com/7.3inch-e-Paper-F.htm](https://www.waveshare.com/7.3inch-e-Paper-F.htm)

What could be the problem?

I added the following to the Waveshare_7_5_3C.ino file

`#include <GxEPD2_7C.h>`
and
`GxEPD2_7C<GxEPD2_730c_GDEY073D46, GxEPD2_730c_GDEY073D46::HEIGHT / 4> display(GxEPD2_730c_GDEY073D46(EPD_CS, EPD_DC, EPD_RST, EPD_BUSY)); // GDEY073D46 800x480 7-color, (N-FPC-001 2021.11.26)
`
from [https://github.com/ZinggJM/GxEPD2](https://github.com/ZinggJM/GxEPD2)

[image]
b.com/ZinggJM/GxEPD2](https://github.com/ZinggJM/GxEPD2)

G6EJD, 2023-05-09:
Why have you set ‘height / 4’ the buffer should be set to half so height/2 ?

paul179, 2023-05-10:
Hello,
I took the value from "GxEPD2_display_selection_added.h" when i change it to 2 i get this error.

`c:/users/tinoo/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\tinoo\AppData\Local\Temp\arduino\sketches\681F510BD3B9E15ABDCC2161FC169940/Waveshare_7_5_3C.ino.elf section .dram0.bss will not fit in region dram0_0_seg
c:/users/tinoo/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/tinoo/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/tinoo/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region dram0_0_seg overflowed by 25352 bytes
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1`

Is my board too small?

Gruß Paul

paul179, 2023-05-14:
Hello,

I got it to work with a "Waveshare ESP32 Driver Board".

I have changed this:

`#include <GxEPD2_7C.h>`

`GxEPD2_7C<GxEPD2_730c_GDEY073D46, GxEPD2_730c_GDEY073D46::HEIGHT / 4> display(GxEPD2_730c_GDEY073D46(EPD_CS, EPD_DC, EPD_RST, EPD_BUSY)); // GDEY073D46 800x480 7-color, (N-FPC-001 2021.11.26)
`

`if (StartWiFi() == WL_CONNECTED && SetupTime() == true) {
    if ((CurrentHour >= WakeupTime && CurrentHour <= SleepTime) || DebugDisplayUpdate) {
      InitialiseDisplay(); // Give screen time to initialise by getting weather data!
      byte Attempts = 1;
      bool RxWeather = false, RxForecast = false;
      WiFiClient client;   // wifi client object
      while ((RxWeather == false || RxForecast == false) && Attempts <= 2) { // Try up-to 2 time for Weather and Forecast data
        if (RxWeather  == false) RxWeather  = obtain_wx_data(client, "weather");
        if (RxForecast == false) RxForecast = obtain_wx_data(client, "forecast");
        Attempts++;
      }
      if (RxWeather && RxForecast) { // Only if received both Weather or Forecast proceed
        StopWiFi(); // Reduces power consumption
        display.setRotation(0);
        int16_t tbx, tby; uint16_t tbw, tbh;
        uint16_t x = ((display.width() - tbw) / 2) - tbx;
        uint16_t y = ((display.height() - tbh) / 2) - tby;
        display.setFullWindow();
        display.firstPage();
        do {
        display.fillScreen(GxEPD_WHITE);
        display.setCursor(x, y);
         DisplayWeather();
        } while (display.nextPage());
      }
    }
  }`

[image]

Gruß Paul

G6EJD, 2023-05-14:
Hello Paul,Well done and it looks good. You could change the temperature colour based on temperature from blue, to green to orange and then red, all sorts of options become available.RegardsDavid

paul179, 2023-05-16:
Hello, if you want you can add the code file  for the Waveshare 7.3inch e-Paper (F).

[https://c.gmx.net/@324822690248523891/Ko5zIBHASqmFRVYmhZPd9w](https://c.gmx.net/@324822690248523891/Ko5zIBHASqmFRVYmhZPd9w)

Greetings Paul

[image]

closed 2023-07-26 by G6EJD