G6EJD/ESP32-e-Paper-Weather-Display #214: Wemos lolin 32+lipo 3,7v 2000mah battery

maik104, 2023-05-18:
Hello,
With a 1.54 inch battery, for example, it lasts about 1.5 days
with 4.2 example several weeks
what can it be? thanks

translated with google

G6EJD, 2023-05-18:
With that display it should last for 6-8 months or more.

What is the exact display type - link?

maik104, 2023-05-18:
Hello, 
https://eckstein-shop.de/Waveshare12C54inch200x200E-InkE-PaperDisplayModulSPIfC3BCrRaspberryPiArduino
rev.2.1

Von: G6EJD
Gesendet: Donnerstag, 18. Mai 2023 17:30
An: G6EJD/ESP32-e-Paper-Weather-Display
Cc: maik104; Author
Betreff: Re: [G6EJD/ESP32-e-Paper-Weather-Display] Wemos lolin 32+lipo 3,7v2000mah battery (Issue #214)

With that display it should last for 6-8 months or more.
What is the exact display type - link?

G6EJD, 2023-05-18:
Which version have you chose:?
//GxEPD2_BW<GxEPD2_154, GxEPD2_154::HEIGHT> display(GxEPD2_154(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEP015OC1 200x200, IL3829, no longer available
//GxEPD2_BW<GxEPD2_154_D67, GxEPD2_154_D67::HEIGHT> display(GxEPD2_154_D67(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEH0154D67 200x200, SSD1681
//GxEPD2_BW<GxEPD2_154_T8, GxEPD2_154_T8::HEIGHT> display(GxEPD2_154_T8(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEW0154T8  152x152, UC8151 (IL0373)
//GxEPD2_BW<GxEPD2_154_M09, GxEPD2_154_M09::HEIGHT> display(GxEPD2_154_M09(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEW0154M09 200x200, JD79653A
//GxEPD2_BW<GxEPD2_154_M10, GxEPD2_154_M10::HEIGHT> display(GxEPD2_154_M10(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEW0154M10 152x152, UC8151D
//GxEPD2_BW<GxEPD2_154_GDEY0154D67, GxEPD2_154_GDEY0154D67::HEIGHT> display(GxEPD2_154_GDEY0154D67(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEY0154D67 200x200, SSD1681, (FPC-B001 20.05.21)

You may have a newer version and unless there are distinct markings to indicate the types then try each one in turn

maik104, 2023-05-18:
GxEPD2_BW<GxEPD2_154_D67, GxEPD2_154_D67::HEIGHT> display(GxEPD2_154_D67(/*CS*/ EPD_CS, /*DC*/ EPD_DC, /*RST*/ EPD_RST, /*BUSY*/ EPD_BUSY)); // New version of 2-Colour display (B/W) GDEH0154D67 or Waveshare 1.54 V2

G6EJD, 2023-05-18:
Have you monitored the serial port to ensure it says entering sleep at the end of an update, just press reset, should take about 5-secs.

closed 2023-05-31 by G6EJD