G6EJD/ESP32-e-Paper-Weather-Display #215: Busy Timeout!

MaillotJaune01, 2023-05-30:
Hello,

the application is running very well for me. Unfortunately, after some time, I receive the message "Busy Timeout!" and the display no longer refreshes.

"
Pop:  0.00
Peri: 2023-06-02 18:00:00
Icon name: 04d
Icon name: 04n
Icon name: 03n
Icon name: 01n
Icon name: 01d
Icon name: 01d
Icon name: 01d
Icon name: 01d
Icon name: 01d
Busy Timeout!
Busy Timeout!
Busy Timeout!
"

What can i do?

G6EJD, 2023-05-30:
There are no known software errors/issues.

So where is your screen located? Is it being illuminated by sunlight, or getting hot/heated or over temperature? These are all factors that can affect the screen and damage its operation. Or it could be failing if it’s an older display and does not have the driver buffers now fitted to the latest types.

G6EJD, 2023-05-31:
The ‘Busy time out’ message comes from the GxEPD2 driver library when it detects the Busy pin has not transitioned within an expected time period, usually this happens when the display fails to complete its refresh cycle. As I asked, the display’s physical environment and UV exposure has a lot to do with performance and failures.

MaillotJaune01, 2023-05-31:
Ok, thank you very much.