G6EJD/ESP32-e-Paper-Weather-Display #216: Can't find common.h

librarymark, 2023-06-01:
No matter where I put the file, the compiler can't find common.h. 
It always says, "Compilation error: common.h: No such file or directory"

What am I missing?

G6EJD, 2023-06-01:
Download the zip file and place in the Library folder or place common.h in the same folder as the main sketch, you will also need lang.h and forecast_record and comm_functions

librarymark, 2023-06-01:
No joy. The file is where it belongs but I still get that error.

G6EJD, 2023-06-01:
Well the correct place is the Library folder in accord with the structure in GitHub. You should download the zip file from GitHub, then start the IDE then go to the Sketch menu and manage libraries, then install a library from zip file and browse to the file just downloaded, then it will be correct.

librarymark, 2023-06-01:
Once I found out where the IDE was expecting things (not where you would expect on my system for some reason) it compiled just fine - thanks!

closed 2023-06-01 by librarymark