G6EJD/ESP32-e-Paper-Weather-Display #220: are there german umlauts?

schroepfer, 2023-07-21:
In OWM weather-messages are no german umlauts.
Example: instead of "Überwiegend bewölkt" ist written on my epaper: "berwiegen bewlkt".

I could not find german umlauts in 'epaper_fonts.h'.

Regards
Jürgen

G6EJD, 2023-07-21:
The fonts come from u8g2Fonts so if there is no font there it can’t be done. Also OWM does not provide a translation service their data is all in English.

schroepfer, 2023-07-21:
I talk from your ESP32 example-code Waveshare_1_54.ino . 
There came the font not from u8g2Fonts.

G6EJD, 2023-07-21:
OK, that version does not support U8g2 fonts, so other than creating a new epaper_font file in German, then sorry there is nothing that can be done.

schroepfer, 2023-07-21:
ok, thanks!