G6EJD/ESP32-e-Paper-Weather-Display #221: ting esp32 to wifi

ArtieRomero, 2023-07-26:


G6EJD, 2023-07-26:
?

ArtieRomero, 2023-07-26:
That must have been due to a typo by me on GitHub when clicking through.
Sorry about that.

But perhaps you can help me with something else?
I made a few of the Weather DIsplays using your GitHub code and flashed the
program to their respective devices to send my immediate family (who are
spread across the U.S.).
In each case, I verified, using my WiFi credentials, that the Weather
Displays were functioning properly and then swapped-out my WiFi credentials
for those of my respective family members.
Obviously, I could not test the Wether Displays after doing that...
The outcome is that two of these worked perfectly upon arrival and two do
not.  The issue is that nothing happens when switched on by the recipient.
It appears that they are not making a WiFi connection.
One of these was sent back to me and I re-flashed it, again using my WiFi
credentials, and it worked perfectly.  So, Again swapped-in the
recipient's WiFi credentials and sent it back, and again, it would not work.

I understand that the ESP32 uses 2.5 GHz and not the 5 GHz of the dual band
system used by WiFi internet transmitters (called access points?).  In any
regard, we are completely sure that the WiFi SSID and passcodes are correct
and that there are connection issues.  Considering that 2 not working out
of 5 is a pretty significant fraction to not work, have you any experience
with this or any suggestions?
My family are not techies and all I can think to do is to ask them to
install Arduino IDE, connect an ESP32 board, and run the Arduino's "WiFi
scan" program to see if it sees their 2.4 GHz signal.

Thanks, Artie

G6EJD, 2023-07-26:
It is almost certain that:
1. The credentials are wrong, usually the SSID name
2. Their routers are 5GHz only units, possible but rare, you need to get the Router brand and type to know. 
3. The units are out of range if their Router.

G6EJD, 2023-07-26:
Odd, I’m using a Netgear RBK752 Orbi Mesh network with 1 base unit and 2 satellites, and the 2.4GHz bs d is visible everywhere. It would seem inconceivable that they have no 2.4GHz units in the house thereby enabling the Orbi to go to single band (5GHz mode), they need to go to the Router settings using the Orbi App and it will report on 2.4GHz devices (5GHz too) and if needs be enable dual band mode.

ArtieRomero, 2023-07-26:
He is using an ORBI RBR50 mesh system.  He tells me that he has a ring
camera that works fine (which I believe uses 2.4 GHz, although the set-up
via a phone app might make a difference for it working?).

1) The SSID and passcode are correct.  I have an actual screenshot of his
unit's label showing this and he has verified the photo that I sent of this
information entered into the weather sketch (and using my SSID.passcode
works with this display).
2) He is sitting only feet away from his router, at his desk, when turning
on the Weather Display (I installed a battery on/off switch to reboot)
3) I read many different descriptions of issues and people not having
issues with ESP32 and or other IoT devices and mesh systems; the literature
is not clear.  Maybe the ORBI specs have changed with different models?  I
will have logon to the Router settings to see what the 2.4 GHz situation
might be.

G6EJD, 2023-07-26:
Each unit produces a dual-band network, so 2.4 and 5GHz then each satellite duplicates that so that you can use a common SSID and password then any device is simply transferred between mesh units depending on their physical location, so any any instance the device can be using the base, a satellite or multiple satellites. I have never known these problems, it’s most likely a router setting that’s causing this issue like a MAC filter thereby only allowing known devices to connect.

ArtieRomero, 2023-07-26:
Yes.  I think it is unlikely that his ORBI would have the 2.4 GHz band
turned off.  I would assume that his Ring camera would not function if this
were so.
But he is not really sophisticated and I think it unlikely that he has
created a "whitelist" in his MAC filter.  I will ask.
When I discover the solution I will report back.

I am also going to dig out a 2 -year old NetGear WiFi extender unit that I
have.  I just checked the online manual and it is supposed to have a fast
data transfer by allowing you to set it to connect in one direction by 5GHz
and the other direction by 2.4 GHz, so I should be able to communicate to
the ESP32 using it.  It may have a more sophisticated method to be
connected "up" to the home WiFi and that might be a backup plan.  A

And I still need to figure out why the other family member cannot get his
Weather Display to work.  This must also be a connection issue, although in
his case he has a standard WiFi transmitter but has used the feature to
separate the 5GHz and 2.4 GHz bands into their own SSIDs.  We are using the
2.4 GHz SSID for the Weather Display.  I'll report back on that, too.

tyler-romero, 2023-08-02:
Thanks for the help! We got it working - not sure what exactly did it, I just recompiled and uploaded the code while on the same wifi network that I was specifying credentials for.

closed 2023-08-02 by G6EJD