G6EJD/ESP32-e-Paper-Weather-Display #222: example doesn't compile with ArduinoJson 6.21.3

schroepfer, 2023-07-29:
...only with 6.9.1 on my MAC M2 and Arduino 2.1.1

`
In file included from /Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:42,
                 from /Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /private/var/folders/lv/6_p55xgx42s8d740lr853dc40000gq/T/.arduinoIDE-unsaved2023629-71776-uwwzm5.bz3a/Waveshare_4_2/Waveshare_4_2.ino:21:
/Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp: In instantiation of 'static T ArduinoJson::V6213PB2::Converter<T, Enable>::fromJson(ArduinoJson::V6213PB2::JsonVariantConst) [with T = char*; Enable = void]':
/Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:53:34:   required from 'typename ArduinoJson::V6213PB2::detail::enable_if<(! ArduinoJson::V6213PB2::detail::ConverterNeedsWriteableRef<T>::value), T>::type ArduinoJson::V6213PB2::detail::VariantRefBase<TDerived>::as() const [with T = char*; TDerived = ArduinoJson::V6213PB2::detail::MemberProxy<ArduinoJson::V6213PB2::detail::ElementProxy<ArduinoJson::V6213PB2::detail::MemberProxy<ArduinoJson::V6213PB2::JsonObject, const char*> >, const char*>; typename ArduinoJson::V6213PB2::detail::enable_if<(! ArduinoJson::V6213PB2::detail::ConverterNeedsWriteableRef<T>::value), T>::type = char*]'
/Users/Juergen-S/Documents/Arduino/libraries/ESP32-e-Paper-Weather-Display/src/common.h:39:72:   required from here
/Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:24:20: error: no matching function for call to 'convertFromJson(ArduinoJson::V6213PB2::JsonVariantConst&, char*&)'
     convertFromJson(src, result);  // Error here? See https://arduinojson.org/v6/unsupported-as/
     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~
/Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:267:13: note: candidate: 'void ArduinoJson::V6213PB2::convertFromJson(ArduinoJson::V6213PB2::JsonVariantConst, String&)' <near match>
 inline void convertFromJson(JsonVariantConst src, ::String& dst) {
             ^~~~~~~~~~~~~~~
/Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:267:13: note:   conversion of argument 2 would be ill-formed:
/Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:24:26: error: cannot bind non-const lvalue reference of type 'String&' to an rvalue of type 'String'
     convertFromJson(src, result);  // Error here? See https://arduinojson.org/v6/unsupported-as/
                          ^~~~~~
In file included from /Users/Juergen-S/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/cores/esp32/Arduino.h:176,
                 from /private/var/folders/lv/6_p55xgx42s8d740lr853dc40000gq/T/arduino/sketches/7789E1265C30736E62B1163BB4E03A9C/sketch/Waveshare_4_2.ino.cpp:1:
/Users/Juergen-S/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/cores/esp32/WString.h:59:9: note:   after user-defined conversion: 'String::String(const char*)'
         String(const char *cstr = "");
         ^~~~~~
In file included from /Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:42,
                 from /Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /private/var/folders/lv/6_p55xgx42s8d740lr853dc40000gq/T/.arduinoIDE-unsaved2023629-71776-uwwzm5.bz3a/Waveshare_4_2/Waveshare_4_2.ino:21:
/Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:283:13: note: candidate: 'void ArduinoJson::V6213PB2::convertFromJson(ArduinoJson::V6213PB2::JsonVariantConst, std::__cxx11::string&)' <near match>
 inline void convertFromJson(JsonVariantConst src, std::string& dst) {
             ^~~~~~~~~~~~~~~
/Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:283:13: note:   conversion of argument 2 would be ill-formed:
/Users/Juergen-S/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:24:26: error: cannot bind non-const lvalue reference of type 'std::__cxx11::string&' {aka 'std::__cxx11::basic_string<char>&'} to an rvalue of type 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'}
     convertFromJson(src, result);  // Error here? See https://arduinojson.org/v6/unsupported-as/
                          ^~~~~~
In file included from /Users/Juergen-S/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/include/c++/8.4.0/string:52,
                 from /Users/Juergen-S/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/include/c++/8.4.0/stdexcept:39,
                 from /Users/Juergen-S/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/include/c++/8.4.0/array:39,
                 from /Users/Juergen-S/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/include/c++/8.4.0/tuple:39,
                 from /Users/Juergen-S/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/include/c++/8.4.0/functional:54,
                 from /Users/Juergen-S/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/cores/esp32/HardwareSerial.h:49,
                 from /Users/Juergen-S/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/cores/esp32/Arduino.h:184,
                 from /private/var/folders/lv/6_p55xgx42s8d740lr853dc40000gq/T/arduino/sketches/7789E1265C30736E62B1163BB4E03A9C/sketch/Waveshare_4_2.ino.cpp:1:
/Users/Juergen-S/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:514:7: note:   after user-defined conversion: 'std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::basic_string(const _CharT*, const _Alloc&) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]'
       basic_string(const _CharT* __s, const _Alloc& __a = _Alloc())
       ^~~~~~~~~~~~
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
  Benutzt: /Users/Juergen-S/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WiFi
  Nicht benutzt: /Users/Juergen-S/Documents/Arduino/libraries/WiFiNINA
exit status 1

Compilation error: exit status 1
`

G6EJD, 2023-07-29:
These errors are all in the Arduino JSON library

closed 2023-07-29 by G6EJD

schroepfer, 2023-07-29:
Ok,
this says, that your program ist ok, and ArduinoJson 6.21.3 defect and 6.9.1 ok?

G6EJD, 2023-07-29:
Exactly