G6EJD/ESP32-e-Paper-Weather-Display #223: support for : dfrobot firebeetle iot esp32+Waveshare_7_5_T7.ino+3,7v lipo

maik104, 2023-08-26:
Hello, can you adapt the code for the dfrobot firebeetle iot esp32 for: Waveshare_7_5_T7.ino, mfg

By the way, the code for 1.54 waveshare display from 18.5.23 (Wemos lolin 32+lipo 3.7v 2000mah battery) still runs perfectly with lipo should I send you the code

G6EJD, 2023-08-26:
There’s no need to modify the code as it will run as-is. The only change needed is the screen wiring and defining the pins you intend to use. SCK, MOSI and MISO and CS remain the same as does DC and so just try out RST and it should all work.

maik104, 2023-08-27:
hello, the code : 
#include "owm_credentials.h"          // See 'owm_credentials' tab and enter your OWM API key and set the Wifi SSID and PASSWORD
#include <ArduinoJson.h>              // https://github.com/bblanchon/ArduinoJson needs version v6 or above
#include <WiFi.h>                     // Built-in
#include "time.h"                     // Built-in
#include <SPI.h>                      // Built-in 
#define  ENABLE_GxEPD2_display 1
#include <GxEPD2_BW.h>
#include <U8g2_for_Adafruit_GFX.h>
#include "epaper_fonts.h"
#include "forecast_record.h"
#include "lang_gr.h"                  // Localisation (German)

#define SCREEN_WIDTH  800             // Set for landscape mode
#define SCREEN_HEIGHT 480

enum alignment {LEFT, RIGHT, CENTER};

// Connections for Firebeetle DFR0478
static const uint8_t EPD_BUSY = 25;  // D2 to EPD BUSY
static const uint8_t EPD_CS   = 5;  // D8 to EPD CS
static const uint8_t EPD_RST  = 26; // D3 to EPD RST
static const uint8_t EPD_DC   = 27; // D4 to EPD DC
static const uint8_t EPD_SCK  = 18; // SCK to EPD CLK
static const uint8_t EPD_MISO = 19; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 23; // MOSI to EPD DIN

GxEPD2_BW<GxEPD2_750_T7, GxEPD2_750_T7::HEIGHT> display(GxEPD2_750_T7(/*CS=D8*/ EPD_CS, /*DC=D3*/ EPD_DC, /*RST=D4*/ EPD_RST, /*BUSY=D2*/ EPD_BUSY));   // B/W display
does not work
the serial monitor says:
15:47:11.085 -> Connecting to: Twietti
15:47:13.394 -> WiFi connected at: 192.168.178.8
15:47:14.387 -> ets Jun  8 2016 00:22:57
15:47:14.387 -> 
15:47:14.387 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
15:47:14.387 -> configsip: 0, SPIWP:0xee
15:47:14.387 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:47:14.387 -> mode:DIO, clock div:1
15:47:14.387 -> load:0x3fff0030,len:1412
15:47:14.433 -> load:0x40078000,len:13400
15:47:14.433 -> load:0x40080400,len:3672
15:47:14.433 -> entry 0x400805f8
15:47:14.771 -> 
15:47:14.771 -> Connecting to: Twietti
15:47:17.056 -> WiFi connected at: 192.168.178.8

and so on

no update of the display

G6EJD, 2023-08-28:
There’s no need to post the source code as there is no error with it and if you have a blurred image then it is a faulty display.