G6EJD/ESP32-e-Paper-Weather-Display #224: Compile error waveshare 7_5_T7

kroon040, 2023-08-27:
Hi,

I tried to compile, and have a waveshare 7_5_T7, with esp32 driverboard. I changed to the right pins , but when compiling I get this error

sketch/src/common_functions.cpp.o: In function `mm_to_inches(float)':
/home/richard/Downloads/ESP32-e-Paper-Weather-Display-master/examples/Waveshare_7_5_T7/src/common_functions.cpp:3: multiple definition of `mm_to_inches(float)'
sketch/common_functions.cpp.o:/home/richard/Downloads/ESP32-e-Paper-Weather-Display-master/examples/Waveshare_7_5_T7/common_functions.cpp:3: first defined here
sketch/src/common_functions.cpp.o: In function `hPa_to_inHg(float)':
common_functions.cpp:(.text._Z11hPa_to_inHgf+0x0): multiple definition of `hPa_to_inHg(float)'
sketch/common_functions.cpp.o:common_functions.cpp:(.text._Z11hPa_to_inHgf+0x0): first defined here
sketch/src/common_functions.cpp.o: In function `JulianDate(int, int, int)':
common_functions.cpp:(.text._Z10JulianDateiii+0x0): multiple definition of `JulianDate(int, int, int)'
sketch/common_functions.cpp.o:common_functions.cpp:(.text._Z10JulianDateiii+0x0): first defined here
sketch/src/common_functions.cpp.o: In function `SumOfPrecip(float*, int)':
common_functions.cpp:(.text._Z11SumOfPrecipPfi+0x0): multiple definition of `SumOfPrecip(float*, int)'
sketch/common_functions.cpp.o:common_functions.cpp:(.text._Z11SumOfPrecipPfi+0x0): first defined here
sketch/src/common_functions.cpp.o: In function `TitleCase(String)':
common_functions.cpp:(.text._Z9TitleCase6String+0x0): multiple definition of `TitleCase(String)'
sketch/common_functions.cpp.o:common_functions.cpp:(.text._Z9TitleCase6String+0x0): first defined here
sketch/src/common_functions.cpp.o: In function `NormalizedMoonPhase(int, int, int)':
common_functions.cpp:(.text._Z19NormalizedMoonPhaseiii+0x0): multiple definition of `NormalizedMoonPhase(int, int, int)'
sketch/common_functions.cpp.o:common_functions.cpp:(.text._Z19NormalizedMoonPhaseiii+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.

G6EJD, 2023-09-03:
You have multiple instances of the libraries, you will need to delete them from the sketch/ folder they should only be in the sketch/src/ folder

closed 2023-09-03 by G6EJD

kroon040, 2023-09-03:
ok.

I did again open from the examples the ino file, with arduino. That I get the error. Or do I need the example to copy to the root folder??

Arduino: 1.8.19 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Waveshare_7_5_T7:29:26: fatal error: epaper_fonts.h: No such file or directory
compilation terminated.
exit status 1
epaper_fonts.h: No such file or directory
Invalid library found in /home/richard/Arduino/libraries/examples: no headers files (.h) found in /home/richard/Arduino/libraries/examples

kroon040, 2023-09-03:
found the issue. Dind not extract and compile the dir in the arduino lib dir.

G6EJD, 2023-09-03:
You need to download the zip file from GitHub and use the IDE to add it using the zip option, then all the files will be placed in the correct location, at the moment your missing the font file.