G6EJD/ESP32-e-Paper-Weather-Display #225: Wemos lolin 32+lipo 3,7v 2000mah battery

maik104, 2023-08-29:
Hello, is it possible in the 1.54 code: instead of displaying the battery sign and the percentage of the battery in the string version? Thanks

G6EJD, 2023-08-29:
Yes it can be done, but first you need to make battery percent a global variable so it can be used anywhere

maik104, 2023-08-29:
Hello, I have already added this:
void drawBattery(int x, int y) {
   uint8_t percentage = 100;
   float voltage = analogRead(35) / 4096.0 * 7.46;
   if (voltage > 1 ) { // Only display if there is a valid reading
     Serial.println("Voltage = " + String(voltage));
     percentage = 2836.9625 * pow(voltage, 4) - 43987.4889 * pow(voltage, 3) + 255233.8134 * pow(voltage, 2) - 656689.7123 * voltage + 632041.7303;
     if (voltage >= 4.20) percentage = 100;
     if (voltage <= 3.50) percentage = 0;
     display.drawRect(x + 15, y - 12, 19, 10, GxEPD_BLACK);
     display.fillRect(x + 34, y - 10, 2, 5, GxEPD_BLACK);
     display.fillRect(x + 17, y - 10, 15 * percentage / 100.0, 6, GxEPD_BLACK);
     drawString(x + 60, y - 11, String(percentage) + "%", RIGHT);
     //drawString(x + 13, y + 5, String(voltage, 2) + "v", CENTER);
   }
}

maik104, 2023-08-29:
and this:
void DisplayHeadingSection() {
  u8g2Fonts.setFont(u8g2_font_helvB08_tf);
  drawString(2, 2, Time_str, LEFT);
  drawString(SCREEN_WIDTH - 2, 0, Date_str, RIGHT);
  drawBattery(2, 0, CENTER);
  //drawString(SCREEN_WIDTH / 2, 0, version, CENTER);
  display.drawLine(0, 12, SCREEN_WIDTH, 12, GxEPD_BLACK);
}
but fail:
Compilation error: too many arguments to function 'void drawBattery(int, int)'

G6EJD, 2023-08-29:
You added too many arguments to the call here:
drawBattery(2, 0, CENTER);
should be:
drawBattery(2, 0);

closed 2023-08-29 by G6EJD