G6EJD/ESP32-e-Paper-Weather-Display #229: Esp32 Waveshare driver board and 7.5" 3 color WFT0583CZ61 not going beyond WiFi Connection

SimonePDA, 2023-09-13:
Hello, I went through the code and you did a great job.
My configuration just dies after the connection to Wifi.

I don't see any other message. I even put some other Serial.print messages trying to find out where the code got stuck with no success.
The code that pulls the weather data does not print any message, so it is never executed.
From some debugging I found out that this 
SetupTime() == true
is never TRUE and the code is stuck there.

If I disable that check, ( SetupTime() != true) the situation doesn't get much better, but the system at least goes to deep sleep.

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.70.82, MASK: 255.255.255.0, GW: 192.168.70.1
WiFi connected at: 192.168.70.82
Busy Timeout!
Entering 1723-secs of sleep time
Awake for : 23.201-secs
Starting deep-sleep period...

The display never activates, but the setup and connection are working as I get it to work with ESPHome Epaper code.

Just ask if you need some more info,

Simone

G6EJD, 2023-09-13:
In my experience it sounds like your TimeZone setting is wrong, what are you using? What happens if you try the example (UK) that I provided.

SimonePDA, 2023-09-14:
OH!
That is interesting!
I am in the  United Arab Emirates, therefore the map of timezones says a strange "<+04>-4"

I spent some more time on the code and found out that the Display Init code at the end had these "new" and "old" communications definitions. I switched to the "old HAT" and everything came to life with HTTPClient Kicking in.
With this setting the display came to life but the screen is somehow blurred and I get a Busy Timeout! at the end of the update, with two or three reverse lines at the sides of the display that also affect the overall image (the image is missing there).

I have played with the display.init() values with no success and only the one with no parameters works, even if with the side lines and blurred result.

So it wasn't the time zone, but the display initialisation.

Now I have to find out how to make it work properly. 
By the way, it is a 3 color and I changed some GxEPD_BLACK into GxEPD_RED and the display changes to RED (cool) 

What do you suggest to fix the SPI dialogue with the display? 

thanks again for your time

Simone

[image]

SimonePDA, 2023-09-15:
The board is a Waveshare ESP32 e-paper universal interface, with the extension flat and the small dual connector PCB. 
The pin assignments are the ones described in the code for the waveshare ESP32 board

static const uint8_t EPD_BUSY = 25;
static const uint8_t EPD_CS   = 15;
static const uint8_t EPD_RST  = 26;
static const uint8_t EPD_DC   = 27;
static const uint8_t EPD_SCK  = 13;
static const uint8_t EPD_MISO = 12; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 14;

I checked the PIN assignment on ESPHome and they are the same:

  gpio_spi_clk_pin: GPIO13
  gpio_spi_mosi_pin: GPIO14
  gpio_cs_pin: GPIO15
  gpio_dc_pin: GPIO27
  gpio_reset_pin: GPIO26
  gpio_busy_pin: GPIO25

The display exact model is Wft0583cz61 that is a 7.5" 3 colors model

As it is a Chinese product I can't rule out it is a defective panel. 

I attach the video of a refresh

https://photos.app.goo.gl/BYnETnNWBtHiW76y7

Simone

G6EJD, 2023-09-15:
OK, the video is very useful. Apart from the alignment of the sunrise/sunset/phase text in the left (easily fixed by moving the x value), it looks OK, but I think there may have been a small increase in x resolution of the display and so there is an misalignment at the far right as the error starts to grow. So try increasing the setting for screen width at the beginning of the programme.

G6EJD, 2023-09-15:
I can see sone 7.5” panels are now 840 pixels wide, so I’m sure that’s what the issue is.

SimonePDA, 2023-09-15:
I will implement the changes right now, but the panel is sold as 800 x 480.
The two bands on the sides that do not display the image (so it looks more and more defective) are still there.
If the panel was properly functioning I think I would see the whole panel with all the parts properly showing.

One question: What are the parameters used when you do a display.init() - without specifying anything? 
I would like to experiment using explicit parameters, but the standard ones don't work ( display.init(115000, true, 2, false); )

G6EJD, 2023-09-15:
It’s possible the panel is defective.
The parameters for the GxEPD driver unit function are all explained in the Github page, many are to get round odd driver boards where Waveshare introduced a reset circuit specifically for the RPi that then ruined the displays for all other systems - they lost their way when they did that, or a lack of experience or both!

SimonePDA, 2023-09-15:
I opened a refund request on aliexpress and I will get another panel (shipping happens fast lately). Same panel same specs. 
The WaveShare board ESP32 looks fine, or at least it is driving the panel somehow. 
The fact that text is a little unfocused is probably part of the damage it got.
I attach macro pics to show better the situation.

For the moment I keep things as they are.

About the Display.init() it works if I use (0, true, 2, false) , quite strange

Simone

[image]
[image]

G6EJD, 2023-09-15:
Now I look at the high resolution images it’s clearly faulty, good luck with the replacement AliExpress are very good with warranty claims.

closed 2023-09-15 by G6EJD