G6EJD/ESP32-e-Paper-Weather-Display #230: Not working with ESP32 S3

mkhan9047, 2023-10-08:
I need urgent help, as I had to change my current MCU from the Wrover module to S3 of the ESP32 lineup. I am using the WaveShare e-paper 4.2 inch. For a long time, I have verified my current pins and connections they are good as reference. But with ESP32 S3 the display is not working anymore. 

@sbonaime  @televator @nicolas-f  and others, please help

G6EJD, 2023-10-08:
You can’t use the WiFi at the same time as the ADC with the S3 you need to swap back to the ESP32

closed 2023-10-08 by G6EJD

mkhan9047, 2023-10-09:
@G6EJD Thanks for your quick response. Can you please tell me in brief why there is a connection between an SPI-based display and ADC with Wifi? Actually, I am in the middle of a project and I am not an expert in this field, but for some reason, I can't replace the ESP32 S3. I must need to get it to work by S3, what is the solution for me?  Please help

G6EJD, 2023-10-09:
It's a ESp32-S3 design change, well as always with the ESP32 you cannot use WiFi with channel-2 of the ADC but on the S3 Espressif chose Channel 2, so you can't use WiFi. You can try disabling the battery monitoring function call, then that won't stop the WiFi from operating which is needed for the Wx data reception along with time services.