G6EJD/ESP32-e-Paper-Weather-Display #236: WiFi.h and possibly time.h & SPI.h are not automatically built in.

Boerny41, 2023-12-30:
Hi,

I'm guessing from the description that WiFi.h is supposed to be automatically included. At least it's not for me and I'd like to ask if you can provide it. The compiler stops there, so I can't check if time.h and SPI.h are actually included, but I'd guess they're missing too.

WiFi.h: No such file or directory
   26 | #include <WiFi.h>              // Built-in
      |          ^~~~~~~~
compilation terminated.

G6EJD, 2023-12-30:
It's almost certain you have not selected an ESP32 board type or your IDE is not the latest (2.2.1 because that includes ESp32 support) or you have not added ESP32 support yo your IDE.

Boerny41, 2023-12-30:
Sorry, my bad. I was indeed using a super outdated version of Arduino Ide. After updating it works. Thanks for the quick help!

closed 2023-12-30 by Boerny41