G6EJD/ESP32-e-Paper-Weather-Display #238: Problem with contrast

Stebe242, 2024-01-29:
I have a pig problem with the contrast. Display 075BN-T7 V2
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/assets/122525247/71d7b886-a43a-46aa-bd05-e0891ce6f722

Stebe242, 2024-01-29:
ok i will try another Display. 
Do you also plan a 7 color version? Or is it possible to use such a diplay?

G6EJD, 2024-01-29:
The text comes from OpenWeatherMap so they must have an issue or are using a slightly different spelling. Easily fixed once you know what they sending. Maybe they have changed ‘clouds’ to ‘Clouds’

G6EJD, 2024-01-29:
Colour version, no plans because it uses too much energy and takes too long to update. It’s easy to do just replace the display and use a new driver object for it and then change the colours in the code as required. Each time a display statement is issued it has a colour parameter all black and white at the present.

Stebe242, 2024-02-07:
Same Problem with new Display and new Controller

G6EJD, 2024-02-07:
Have you tried the GxEPD2 gfx example in that library?
Your dependency is the library author providing support for a display that has been updated.

Stebe242, 2024-02-11:
The Problem was the little switch 

[image]

closed 2024-02-13 by G6EJD