G6EJD/ESP32-e-Paper-Weather-Display #240: Display stops updating....

njordan77, 2024-01-30:
I'm using your great project with Firebeetle2 and 7,5inch Displays. 800x  and 640x versions.
It really works great and it seems that also my battery lasts much longer than any deep sleep i tried myself before, big big thanks to the developer.

Only observation that brings a little shadow is that for example the update stopped at 6pm and now we have 8:30pm without any new update.....from the openweathermap i see that data can be reached.

Any idea why? i use your core code also for a "smart home" info display, and its the same...

at 9pm it worked/updated again....
anything i could to do make this not happen. What i always ask myself how "congested" Wifi is, so that maybe its too many packets and resulting in ESP32 not able to reach any channel to transmit data.....but here i'm to unexperienced to have an expert view.

G6EJD, 2024-01-30:
The most likely reason is having too many devices on your router,  most are limited to 32 maximum. Ideally monitor the serial port on or around 18:00 so you can see why the update is failing. As the ESP32 is asleep and does not maintain a network connection it is highly likely when it wakes the connection fails. Perhaps consider what changes at or around 18:00 as that will give a clue. But nothing beats monitoring the serial port to find out the reason.

njordan77, 2024-01-30:
Thanks for the great response time. highly appreciated. I have right now 6 displays running, 2 weather app, the rest my openhab smart home info panels....all your core code (which is really way more advanced than what i was able to do myself, said again).

maybe really its too many parallel wifi channels towards the FritzBox Router... i never heard of this 32 nodes limitation, but i guess you mean parallel active connections, as i most likely do exceed easily this value. Raspi, 3 Notebooks, 4 mobiel phones, 2 tablets, heating, AC, Video DoorBell, arround 20 ESPs, Lora-Gateway, Telsa, .......

G6EJD, 2024-01-30:
You could have 100 devices registered but there is always a limit to the number of IP addresses that can be assigned on your LAN. 32 simultaneous connections is a typical limit imposed by Routers. A typical LAN address is eg 192.168.0.1 and the last field has a maximum range of 253, where at least one address is retained for the Router, let’s say 192.168.0.1, that leaves 253 spare, 0 is never used. But the open source Router firmware used by most suppliers does have a limit and 32 is typical. Even so try monitoring the serial port as it wakes at 18:00

njordan77, 2024-01-30:
sure, will go serial to get to the reason for not updating. its not the topic, but 32 seems rather low but depends of what is ment by the firmware. concurrent active connections. maybe. im familiar with ip subnetting, so the limit is arround 253 removing router ip, broadcast,...maybe i should also spread the wake up time of these 6 displays. as right now they all wake up exactlythe same 30minutes timeframes...so a shift of 5mins per display would remove some stress for the tiny ESPs.

G6EJD, 2024-01-30:
The Fritzbox has a limit of 40 simultaneous connections, sounds like you’re hitting that limit. 
All Routers use the same Router source code developed under the gnu licence, I’ve not seen any brand that does not use the same source code. 
Spreading the wake-up time will help ensure enough spare capacity as they are not awake for long.

njordan77, 2024-01-30:
wow, great insight. was googling a little but did not find any hard facts. also interesting that there is a GNU level core software used in any/many routers.

Regarding the spread, if i want a 30mins routine, i guess it would not help to change the 30 to 25 in SleepDuration.

It would be here (?)
long SleepTimer = (SleepDuration * 60 - ((CurrentMin % SleepDuration) * 60 + CurrentSec));

G6EJD, 2024-01-30:
It’s simply the sleep duration in minutes, so set it accordingly plus an adjustment for partial minutes and seconds for accuracy

closed 2024-02-13 by G6EJD