G6EJD/ESP32-e-Paper-Weather-Display #242: No picture

Trycilly, 2024-02-07:
This is my first try doing something like this. Which means my issue is pretty easy to solve I just have no Idea how to work with an Arduino.
This is my 
I'm using the waveshare esp32 
And the waveshare 7.5 e-paper display
I changed all my owm_credentials and used this in the code

static const uint8_t EPD_BUSY = 25;
static const uint8_t EPD_CS   = 15;
static const uint8_t EPD_RST  = 26; 
static const uint8_t EPD_DC   = 27; 
static const uint8_t EPD_SCK  = 13;
static const uint8_t EPD_MISO = 12; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 14;

but I'm not getting an output on

G6EJD, 2024-02-07:
You've got your serial port set to 9600baud, it should be 115200.

If you have a new style display from Waveshare, then you need to apply 3.3v to the PWR pin.

Are you getting weather data on the serial port?

Trycilly, 2024-02-07:
Yes I have 3.3v appield to the PWR pin.

Also yes I 'm getting weather data on the serial port after changing it to 115200

G6EJD, 2024-02-07:
Have you selected the correct display driver? It could be:
GxEPD2_BW<GxEPD2_750, GxEPD2_750::HEIGHT> display(GxEPD2_750(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY));   // B/W display
Or:
//#define GxEPD2_DRIVER_CLASS GxEPD2_750     // GDEW075T8   640x384, UC8159c (IL0371), (WF0583CZ09)
//#define GxEPD2_DRIVER_CLASS GxEPD2_750_T7  // GDEW075T7   800x480, EK79655 (GD7965), (WFT0583CZ61)
//#define GxEPD2_DRIVER_CLASS GxEPD2_750_GDEY075T7  // GDEY075T7  800x480, UC8179 (GD7965), (FPC-C001 20.08.20)

Trycilly, 2024-02-07:
I used this one

GxEPD2_BW<GxEPD2_750_T7, GxEPD2_750_T7::HEIGHT> display(GxEPD2_750_T7(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY));

G6EJD, 2024-02-07:
This is a GxEPD2 library issue but you have to go through the display object lines to see which one works. It could be a driver does not yet exist for the display.

Trycilly, 2024-02-07:
How can I check if I there is a driver for the display? And if I have it. I've seen it work with other people so could be I just did not download the driver?
Waveshare 7.5 Inch E-Paper Raw Display Panel V2 800x480  This is the one I got. So I have to use the 7_5_T7 version. RIght?

Trycilly, 2024-02-07:
I did select that one 
[image]