G6EJD/ESP32-e-Paper-Weather-Display #243: No Display

colussim, 2024-02-09:
Hi ,
i use a Waveshare 7.5 inch E-Paper Display Hat Module V2 Kit 800x480.I tested with your example code Waveshare_7_5, but nothing appears on the screen. I can see the measurements on the console.I tested with the example code this waveshare: https://github.com/waveshareteam/e-Paper/blob/master/Arduino/epd7in5b_V2/epd7in5b_V2.ino , I have something on the screen. Any idea what the problem is?

Thanks

Emmanuel

G6EJD, 2024-02-09:
Have you applied a permanent 3.3v to the PWR pin of the display?

If so, then this is a GxEPD2 library issue, there's nothing wrong with the Wx code.

closed 2024-02-09 by G6EJD

G6EJD, 2024-02-09:
It will be because the GxEPD2 library does not yet support that display.

colussim, 2024-02-09:
Yes, I have a voltage of 3.3v on the PWR pin.What type of E-Paper Display do you recommend?

G6EJD, 2024-02-09:
If you want to upgrade the version, you can choose one of the following ways:
Update the program to the newest one (Arduino program only supports 7.5inch e-Paper, 7.5inch e-Paper (B), 5.83inch e-Paper, 5.83inch e-Paper (B), 2.9inch e-Paper (D), 2.13inch e-Paper (D)).
Connect PWR pin to 3.3V.
The empty pad on the back side is soldered with a 47K (or similar) resistor.
E-Paper Driver HAT11.jpg

See on the Wavshare v2 HAT documentation

G6EJD, 2024-02-09:
https://forum.arduino.cc/t/waveshare-7-5in-e-ink-screen-not-working-with-esp32/1219996

colussim, 2024-02-10:
Thanks,
Emmanuel