G6EJD/ESP32-e-Paper-Weather-Display #248: Update

mrmike8887, 2024-04-16:
Will there be a quick how-to for the new api update?

G6EJD, 2024-04-16:
Not sure what you mean, has the WU API been updated? If so that will invalidate many commercial devices that continue to upload to WU. 

Unless you pay WU $500/month you can’t upload or download any data without their API key. If you maintained a Personal Weather Station (PWS) across the IBM takeover of WU, then you can continue to upload data but not download it. You can view data on the WU dashboard.

mrmike8887, 2024-04-16:
The openweathermap API.  I have WU, but I followed your system with these guys.

Dear OpenWeather user,

 We would like to inform you about our plans to complete the migration from One Call 2.5 to One Call 3.0 that started 2 years ago. In line with this, access to One Call 2.5 will be finally closed in June 2024.

G6EJD, 2024-04-16:
Ok, you’re confusing me because this is under the Weatherunderground code repository and not the Openweathermap variant. 
I haven’t seen any correspondence from OWM but I’ll look into the changes.

G6EJD, 2024-04-16:
A cursory check on the differences shows there is compatibility between 2.5 and 3.0 API’s with the key difference being with 3.0 calls above 2000/day will be charged to a register payment method, this will undoubtedly require a new API key and re compilation accordingly. At a typical rate of 24/day calls those codes are well below the limit imposed, but anyone who chooses to call faster will pay for the privilege!

I conclude no code changes required snd all uses will have to register with payment details to get the new Key.

closed 2024-04-16 by G6EJD