G6EJD/ESP32-e-Paper-Weather-Display #254: OWM is moving away from using city-location API to fetch weather

ArtieRomero, 2024-04-24:
I have been communicating with OWM about my Weather Displays in the USA having difficulties with the API call.  After a long period of working fine, the Weather Displays suddenly stopped working and I found that making a small change in the city,state formatting in the Weather Display's code for making the API call fixed the problem.  I passed my problem and resolution to OWM for their information.

OWM responded to say that API requests by city name are deprecated.  See their response, below:

"Please use [Geocoder API](https://openweathermap.org/api/geocoding-api) if you need to automatically convert city names and zip codes to geo coordinates and the other way around. 

Please note that [API requests by city name](https://openweathermap.org/current#name), [zip-codes](https://openweathermap.org/current#zip) and [city id](https://openweathermap.org/current#cityid) have been deprecated. Although they are still available for use, bug fixing and updates are no longer available for this functionality."

This message is also on the OWM webpage:   https://openweathermap.org/current#name

 Would it be too much trouble to modify the Weather DIsplay's code to use Lat and Long to make the API call?
Thanks

G6EJD, 2024-04-24:
OK, I’ll do that, keep a look out. I’ll need to consider how to do it as easily as possible.

ArtieRomero, 2024-04-24:
I am just an amateur (a chemist, actually) and experimented, replacing
"Melksham,UK" with a made-up example lat=38.50,long=-90.63 in your "Example
API call for weather data"  in the owm_credentials page of code and this
worked when entering the string as a URL query at OWM.

My initial attempt was to try to use the lat and long information in the
weather program where the code asks for city and country in the
owm_credentials tab.

However, the OWM API formatting for city,country based API calls
incorporates a preceding "q=" before the city,country.
If it was not for that, I think that one could simply replace "String City
=" with "String Latitude = " and replace "String Country = " with "String
Latitude = "
However I could not locate the code that assembles these strings and makes
the API call.  Is it hiding in one of the included libraries?

Note the inclusion of the text (yellow highlighting) for city,country,
below:
Lat and long-based API call format:
https://api.openweathermap.org/data/2.5/weather?lat={lat}&lon={lon}&appid={API
key} <https://home.openweathermap.org/api_keys>

e.g.
https://api.openweathermap.org/data/2.5/weather?lat=44.34&lon=10.99&appid={API
key} <https://home.openweathermap.org/api_keys>

City and state-based API call format:
https://api.openweathermap.org/data/2.5/weather?q={city name},{country code}
&appid={API key} <https://home.openweathermap.org/api_keys>

e.g.    https://api.openweathermap.org/data/2.5/weather?q=London,uk&appid={API
key} <https://home.openweathermap.org/api_keys>

G6EJD, 2024-04-24:
I’ll give your idea a try, the API call is constructed in the folder ’src’ then common.h and the function ‘obtain_wx_data‘
You’ll find the files in the library folder.

ArtieRomero, 2024-04-24:
I kept digging and found the SRC folder that you mention and see the code there.  
I will play with this tonight.  These changes should work.
And users can easily obtain their locations' lat and long from OWM which appears to be keeping that service.

ArtieRomero, 2024-04-25:
David,

I managed to get the Lat and Lon inputs to work in the code to replace City
and Country.

Below are the changes I made, shown with my location in the town of
Wildwood with its Lat and Lon coordinates.

In the owm_credentials.h tab, I replaced the "String Country =" with these
two additional lines to hold the Lat and Lon information.  Although the
"String City =" information is no longer needed to make the OWM API call,
this input is still used to place the location name at the top of the
e-Paper display, so it is kept.

String City           = "Wildwood";
String Lat             = "38.68";
String Lon           = "-90.20";

I also went into the
sketchbook/libraries/ESP32-e-Paper-Weather-Display-master/src folder and
used Notepad to open and edit the common.h  file.  In this file I  edited
this preexisting line

String uri = "/data/2.5/" + RequestType + "?q=" + City + "," + Country +
"&APPID=" + apikey + "&mode=json&units=" + units + "&lang=" + Language;

And changed it to:

String uri = "/data/2.5/" + RequestType + "?" + "lat=" + Lat + "&lon=" +
Lon + "&APPID=" + apikey + "&mode=json&units=" + units + "&lang=" +
Language;

And then saved and exited the common.h file.

This is all it took.
For readers, one can obtain their Lat and Lon coordinates from OWM, but it
is just as convenient and more precise to use Google Maps on a computer or
smartphone: find the location on the map and then drop a "pin" on the
location.  On a smartphone this is done by pressing the blue icon in the
lower left, labeled "Explore" and then touching the location on the map.
After dropping the pin, scroll to the bottom of the screen and see the
latitude and longitude for that location.  On a Windows PC running Google
Maps, select the icon in the lower right that looks like a target and then
select the location on the map.
Cheers

G6EJD, 2024-04-25:
I have now updated the common.h and all of the examples to include LAT LON variables.

Many thanks for your help.

The next change will be API 3.0 introduction in June 24, but at the moment the changes are scant, I'm hoping it will be 2.5 changes to 3.0, but it could be just 3, who knows!

G6EJD, 2024-04-25:
Closed now as completed.

closed 2024-04-25 by G6EJD

ArtieRomero, 2024-04-25:
Thanks David.
It is a great project!